code-iai/iai_kinect2 #394: Kinect2Bridge::start] Initialization failed Failed to load nodelet '/kinect2_bridge` of type `kinect2_bridge/kinect2_bridge_nodelet` to manager `kinect2'

peddaramesh, 2017-03-29:
ritchie@bell-lab:~/kinect_xbox$ roslaunch kinect2_bridge kinect2_bridge.launch 
... logging to /home/ritchie/.ros/log/4d26ad4c-1495-11e7-9219-5404a61c4105/roslaunch-bell-lab-12817.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://bell-lab:33074/

SUMMARY
========

PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/ritchie/kin...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 12.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: False
 * /kinect2_bridge/queue_size: 5
 * /kinect2_bridge/reg_device: -1
 * /kinect2_bridge/reg_method: default
 * /kinect2_bridge/sensor: 
 * /kinect2_bridge/use_png: False
 * /kinect2_bridge/worker_threads: 4
 * /kinect2_points_xyzrgb_hd/queue_size: 5
 * /kinect2_points_xyzrgb_qhd/queue_size: 5
 * /kinect2_points_xyzrgb_sd/queue_size: 5
 * /rosdistro: kinetic
 * /rosversion: 1.12.6

NODES
  /
    kinect2 (nodelet/nodelet)
    kinect2_bridge (nodelet/nodelet)
    kinect2_points_xyzrgb_hd (nodelet/nodelet)
    kinect2_points_xyzrgb_qhd (nodelet/nodelet)
    kinect2_points_xyzrgb_sd (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [12828]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 4d26ad4c-1495-11e7-9219-5404a61c4105
process[rosout-1]: started with pid [12841]
started core service [/rosout]
process[kinect2-2]: started with pid [12858]
process[kinect2_bridge-3]: started with pid [12859]
process[kinect2_points_xyzrgb_sd-4]: started with pid [12860]
process[kinect2_points_xyzrgb_qhd-5]: started with pid [12862]
[ INFO] [1490801714.257062983]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
process[kinect2_points_xyzrgb_hd-6]: started with pid [12880]
[ INFO] [1490801714.284545093]: waitForService: Service [/kinect2/load_nodelet] has not been advertised, waiting...
[ INFO] [1490801714.296967881]: Initializing nodelet with 4 worker threads.
[ INFO] [1490801714.310961870]: waitForService: Service [/kinect2/load_nodelet] is now available.
[ INFO] [1490801714.449682537]: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/ritchie/kinect_xbox/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: default
     depth_device: -1
       reg_method: default
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 5
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Error] [OpenCLDepthPacketProcessorImpl] could not find any suitable device
[Error] [OpenCLAllocator] b->buffer = cl::Buffer(context, CL_MEM_WRITE_ONLY | CL_MEM_ALLOC_HOST_PTR, size, NULL, &err): -34
[Error] [OpenCLAllocator] b->buffer = cl::Buffer(context, CL_MEM_WRITE_ONLY | CL_MEM_ALLOC_HOST_PTR, size, NULL, &err): -34
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_39
libva info: va_openDriver() returns 0
[Info] [VaapiRgbPacketProcessorImpl] driver: Intel i965 driver for Intel(R) Sandybridge Mobile - 1.7.0
[Error] [VaapiRgbPacketProcessorImpl] vaQueryConfigEntrypoints(display, VAProfileJPEGBaseline, entrypoints, &num_entrypoints): the requested VAProfile is not supported
[Error] [OpenCLAllocator] b->buffer = cl::Buffer(context, CL_MEM_READ_ONLY | CL_MEM_ALLOC_HOST_PTR, size, NULL, &err): -34
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Error] [Freenect2Impl] failed to open Kinect v2: @4:3 LIBUSB_ERROR_ACCESS Access denied (insufficient permissions)
[Info] [Freenect2Impl] found 0 devices
[ERROR] [1490801714.550538856]: [Kinect2Bridge::initDevice] no Kinect2 devices found!
[Error] [OpenCLAllocator] queue.enqueueUnmapMemObject(b->buffer, b->data, NULL, &event): -36
[Error] [OpenCLAllocator] queue.enqueueUnmapMemObject(b->buffer, b->data, NULL, &event): -36
[Error] [OpenCLAllocator] queue.enqueueUnmapMemObject(b->buffer, b->data, NULL, &event): -36
[ERROR] [1490801714.550796605]: [Kinect2Bridge::start] Initialization failed!
[FATAL] [1490801714.621521076]: Failed to load nodelet '/kinect2_bridge` of type `kinect2_bridge/kinect2_bridge_nodelet` to manager `kinect2'
[kinect2_bridge-3] process has died [pid 12859, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/home/ritchie/.ros/log/4d26ad4c-1495-11e7-9219-5404a61c4105/kinect2_bridge-3.log].
log file: /home/ritchie/.ros/log/4d26ad4c-1495-11e7-9219-5404a61c4105/kinect2_bridge-3*.log
[kinect2_bridge-3] restarting process
process[kinect2_bridge-3]: started with pid [13043]
[ INFO] [1490801715.006810774]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1490801715.069262488]: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/ritchie/kinect_xbox/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: default
     depth_device: -1
       reg_method: default
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 5
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Error] [OpenCLDepthPacketProcessorImpl] could not find any suitable device
[Error] [OpenCLAllocator] b->buffer = cl::Buffer(context, CL_MEM_WRITE_ONLY | CL_MEM_ALLOC_HOST_PTR, size, NULL, &err): -34
[Error] [OpenCLAllocator] b->buffer = cl::Buffer(context, CL_MEM_WRITE_ONLY | CL_MEM_ALLOC_HOST_PTR, size, NULL, &err): -34
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_39
libva info: va_openDriver() returns 0
[Info] [VaapiRgbPacketProcessorImpl] driver: Intel i965 driver for Intel(R) Sandybridge Mobile - 1.7.0
[Error] [VaapiRgbPacketProcessorImpl] vaQueryConfigEntrypoints(display, VAProfileJPEGBaseline, entrypoints, &num_entrypoints): the requested VAProfile is not supported
[Error] [OpenCLAllocator] b->buffer = cl::Buffer(context, CL_MEM_READ_ONLY | CL_MEM_ALLOC_HOST_PTR, size, NULL, &err): -34
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Error] [Freenect2Impl] failed to open Kinect v2: @4:3 LIBUSB_ERROR_ACCESS Access denied (insufficient permissions)
[Info] [Freenect2Impl] found 0 devices
[ERROR] [1490801715.071153174]: [Kinect2Bridge::initDevice] no Kinect2 devices found!
[Error] [OpenCLAllocator] queue.enqueueUnmapMemObject(b->buffer, b->data, NULL, &event): -36
[Error] [OpenCLAllocator] queue.enqueueUnmapMemObject(b->buffer, b->data, NULL, &event): -36
[Error] [OpenCLAllocator] queue.enqueueUnmapMemObject(b->buffer, b->data, NULL, &event): -36
[ERROR] [1490801715.071333789]: [Kinect2Bridge::start] Initialization failed!
[FATAL] [1490801715.135059273]: Failed to load nodelet '/kinect2_bridge` of type `kinect2_bridge/kinect2_bridge_nodelet` to manager `kinect2'
[kinect2_bridge-3] process has died [pid 13043, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/home/ritchie/.ros/log/4d26ad4c-1495-11e7-9219-5404a61c4105/kinect2_bridge-3.log].
log file: /home/ritchie/.ros/log/4d26ad4c-1495-11e7-9219-5404a61c4105/kinect2_bridge-3*.log

peddaramesh, 2017-03-29:
I am getting this kind of error.Could any one please help me
Thanks in advance

TheTesla, 2017-05-06:
I am getting this kind of error, but often/sporadically. Maybe a race condition?

wiedemeyer, 2017-05-23:
Looks like something with you OpenCL setup is wrong. Either fix it or use CPU/OpenGL/CUDA as depth processor and CPU for registration.

closed 2017-05-23 by wiedemeyer