code-iai/iai_kinect2 #398: Motion capture markers black center on IR camera

easypeet, 2017-03-31:
I just introduced motion capture markers to my system (small spheres with retroreflective material). These markers are highly reflective in a IR image. When I run kinect2_bridge and visualize the IR topic (/kinect2/sd/image_ir) in rviz I get the following image: 
[image]

As you can see there are three markers in the image, one on top of the robot and two on the table. The problem is that the centers of the markers become black. This should not happen and I do not understand exactly why. My only suggestion is that there are some saturation limit on the intensity level of the pixel values. I hope you can help me out.

JimmyDaSilva, 2017-03-31:
I can confirm I have the same issue here. And this is kind of a problem for me as I use those markers for autocalibration of the extrinsics.
It seems there is a saturation somewhere indeed because this happens when the balls get closer to the sensor.
This does not happen with openni when using either the Xtion or the Kinect1

[image]
[image]

easypeet, 2017-03-31:
@JimmyDaSilva Good that I am not alone. Hopefully there are a quick fix to this. Openni also uses libfreenect/libfreenect2 if I am not mistaken? In that case the issue must lie in iai_kinect2.

JimmyDaSilva, 2017-03-31:
I meant openni_camera which does not use libfreenect2. openni_camera or openni2_camera are like the equivalent of libfreenect2+iai_kinect2 for the Asus Xtion and KinectV1

easypeet, 2017-03-31:
Hmm, then libfreenect2 should also be investigated.

wiedemeyer, 2017-05-23:
It will happen, if something is overexposed. I don't think there is a fix for it, because the exposure time cannot be changed, the device does it automatically. But you could check if the raw values from libfreenect2 are zero too, just to make sure.