code-iai/iai_kinect2 #399: roslaunch kinect2_bridge kinect2_brdige.launch crashes

buddiman2, 2017-03-31:
**Hello,

I cannot run** 
roslaunch kinect2_bridge kinect2_brdige.launch

**after that get this:**
... logging to /home/xiao/.ros/log/07720274-1625-11e7-b796-4c0f6efd8dbf/roslaunch-xiao-VPCCA1S1E-14463.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://xiao-VPCCA1S1E:44300/

SUMMARY
========

PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/xiao/catkin...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 12.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: False
 * /kinect2_bridge/queue_size: 5
 * /kinect2_bridge/reg_device: -1
 * /kinect2_bridge/reg_method: default
 * /kinect2_bridge/sensor: 
 * /kinect2_bridge/use_png: False
 * /kinect2_bridge/worker_threads: 4
 * /kinect2_points_xyzrgb_hd/queue_size: 5
 * /kinect2_points_xyzrgb_qhd/queue_size: 5
 * /kinect2_points_xyzrgb_sd/queue_size: 5
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /
    kinect2 (nodelet/nodelet)
    kinect2_bridge (nodelet/nodelet)
    kinect2_points_xyzrgb_hd (nodelet/nodelet)
    kinect2_points_xyzrgb_qhd (nodelet/nodelet)
    kinect2_points_xyzrgb_sd (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [14475]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 07720274-1625-11e7-b796-4c0f6efd8dbf
process[rosout-1]: started with pid [14488]
started core service [/rosout]
process[kinect2-2]: started with pid [14504]
process[kinect2_bridge-3]: started with pid [14506]
process[kinect2_points_xyzrgb_sd-4]: started with pid [14507]
process[kinect2_points_xyzrgb_qhd-5]: started with pid [14508]
process[kinect2_points_xyzrgb_hd-6]: started with pid [14511]
[ INFO] [1490973396.012543947]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1490973396.031824906]: waitForService: Service [/kinect2/load_nodelet] has not been advertised, waiting...
[ INFO] [1490973396.065199452]: Initializing nodelet with 4 worker threads.
[ INFO] [1490973396.075804329]: waitForService: Service [/kinect2/load_nodelet] is now available.
[ INFO] [1490973396.171021180]: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
[... 151 lines cut ...]
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [Kinect2Bridge::initDevice] device serial: 005691753647
[ INFO] [Kinect2Bridge::initDevice] device firmware: 4.0.3911.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ERROR] [Kinect2Bridge::initRegistration] CPU registration is not available!
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[ERROR] [Kinect2Bridge::start] Initialization failed!
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing

**i am using Advanced Micro Devices, Inc. [AMD/ATI] Seymour [Radeon HD 6400M/7400M Series] GPU and followed the libfreenect2 install tutorial.
./bin/Protonect works but just the other thing doesnt work :( please help**

LabJeff123, 2017-04-08:
I'm having the similar issue when using CPU as registration... I don't have GPU so I'm not using opencl. I did a "rosrun kinect2_bridge kinect2_bridge _depth_method:=cpu _reg_method:=cpu", and here is the output:
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/anca/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: cpu
     depth_device: -1
       reg_method: cpu
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_39
libva info: va_openDriver() returns 0
[Info] [VaapiRgbPacketProcessorImpl] driver: Intel i965 driver for Intel(R) Sandybridge Desktop - 1.7.0
[Error] [VaapiRgbPacketProcessorImpl] vaQueryConfigEntrypoints(display, VAProfileJPEGBaseline, entrypoints, &num_entrypoints): the requested VAProfile is not supported
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:5 with serial 006400461647
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 006400461647 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [Kinect2Bridge::initDevice] device serial: 006400461647
[ INFO] [Kinect2Bridge::initDevice] device firmware: 4.0.3916.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ERROR] [Kinect2Bridge::initRegistration] CPU registration is not available!
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[ERROR] [Kinect2Bridge::start] Initialization failed!
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing

Protonect works for opengl and cpu. Please help!!!

wiedemeyer, 2017-05-23:
This happens if the kinect2_registration package did not compiled the CPU registration. Please check the cmake output, but probably Eigen was not found.