code-iai/iai_kinect2 #401: " catkin_make -DCMAKE_BUILD_TYPE="Release" "failed

wangxiansheng0122, 2017-04-10:
I follow the installion step by step .But some error happened when I running the " catkin_make -DCMAKE_BUILD_TYPE="Release" ". What should I do?

Base path: /home/wjx/catkin_ws
Source space: /home/wjx/catkin_ws/src
Build space: /home/wjx/catkin_ws/build
Devel space: /home/wjx/catkin_ws/devel
Install space: /home/wjx/catkin_ws/install
####
#### Running command: "cmake /home/wjx/catkin_ws/src -DCMAKE_BUILD_TYPE=Release -DCATKIN_DEVEL_PREFIX=/home/wjx/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/wjx/catkin_ws/install -G Unix Makefiles" in "/home/wjx/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/wjx/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/wjx/catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/wjx/catkin_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/wjx/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.18
-- BUILD_SHARED_LIBS is on
/opt/ros/indigo/share/catkin/cmake/em/order_packages.cmake.em:23: error: <type 'exceptions.RuntimeError'>: Multiple packages found with the same name "joy":
- share/joy
- share/joystick_drivers-indigo-devel/joy
Traceback (most recent call last):
  File "/usr/bin/empy", line 3276, in <module>
    if __name__ == '__main__': main()
  File "/usr/bin/empy", line 3274, in main
    invoke(sys.argv[1:])
  File "/usr/bin/empy", line 3257, in invoke
    interpreter.wrap(interpreter.file, (file, name))
  File "/usr/bin/empy", line 2262, in wrap
    self.fail(e)
  File "/usr/bin/empy", line 2253, in wrap
    callable(*args)
  File "/usr/bin/empy", line 2326, in file
    self.safe(scanner, done, locals)
  File "/usr/bin/empy", line 2368, in safe
    self.parse(scanner, locals)
  File "/usr/bin/empy", line 2388, in parse
    token.run(self, locals)
  File "/usr/bin/empy", line 1403, in run
    interpreter.execute(self.code, locals)
  File "/usr/bin/empy", line 2565, in execute
    exec(statements, self.globals)
  File "<string>", line 17, in <module>
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/topological_order.py", line 114, in topological_order
    for path, package in find_packages(space).items():
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/packages.py", line 92, in find_packages
    raise RuntimeError('\n'.join(duplicates))
RuntimeError: Multiple packages found with the same name "joy":
- share/joy
- share/joystick_drivers-indigo-devel/joy
CMake Error at /opt/ros/indigo/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/home/wjx/catkin_ws/build/catkin_generated/env_cached.sh
  "/usr/bin/python" "/usr/bin/empy" "--raw-errors" "-F"
  "/home/wjx/catkin_ws/build/catkin_generated/order_packages.py" "-o"
  "/home/wjx/catkin_ws/build/catkin_generated/order_packages.cmake"
  "/opt/ros/indigo/share/catkin/cmake/em/order_packages.cmake.em") returned
  error code 1
Call Stack (most recent call first):
  /opt/ros/indigo/share/catkin/cmake/em_expand.cmake:25 (safe_execute_process)
  /opt/ros/indigo/share/catkin/cmake/catkin_workspace.cmake:35 (em_expand)
  CMakeLists.txt:63 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/wjx/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/wjx/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

wangxiansheng0122, 2017-04-11:
Please help me !

wangxiansheng0122, 2017-04-11:
@wiedemeyer

wiedemeyer, 2017-05-23:
This has nothing to do with this package. Maybe something with your system configuration is wrong.

closed 2017-05-23 by wiedemeyer