code-iai/iai_kinect2 #402: Error occurs after run catkin_make -DCMAKE_BUILD_TYPE="Release"

la9881275, 2017-04-20:
The result is as follows:
……
Built target kinect2_bridge
[100%] Building CXX object iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/src/viewer.cpp.o
In file included from /usr/include/pcl-1.7/pcl/visualization/cloud_viewer.h:39:0,
                 from /home/zwp/catkin_ws/src/iai_kinect2/kinect2_viewer/src/viewer.cpp:32:
/usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:283:88: error: macro "PCL_DEPRECATED" requires 2 arguments, but only 1 given
         "addCoordinateSystem (scale, id, viewport) with id a unique string identifier.")
                                                                                        ^
/usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:304:97: error: macro "PCL_DEPRECATED" requires 2 arguments, but only 1 given
         "addCoordinateSystem (scale, x, y, z, id, viewport) with id a unique string identifier.")
                                                                                                 ^
/usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:327:91: error: macro "PCL_DEPRECATED" requires 2 arguments, but only 1 given
         "addCoordinateSystem (scale, t, id, viewport) with id a unique string identifier.")
                                                                                           ^
/usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:374:81: error: macro "PCL_DEPRECATED" requires 2 arguments, but only 1 given
         "addCoordinateSystem (id, viewport) with id a unique string identifier.")
                                                                                 ^
/usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:281:9: error: ‘PCL_DEPRECATED’ does not name a type
         PCL_DEPRECATED (
         ^
/usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:302:9: error: ‘PCL_DEPRECATED’ does not name a type
         PCL_DEPRECATED (
         ^
/usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:325:9: error: ‘PCL_DEPRECATED’ does not name a type
         PCL_DEPRECATED (
         ^
/usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:372:9: error: ‘PCL_DEPRECATED’ does not name a type
         PCL_DEPRECATED (
         ^
make[2]: *** [iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/src/viewer.cpp.o] Error 1
make[1]: *** [iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/all] Error 2
make: *** [all] Error 2

Jon-218, 2017-05-01:
The same issue in my workspace，

Jon-218, 2017-05-01:
The same issue in my workspace，did you fixed it？

la9881275, 2017-05-02:
@Jon-218 Not yet. I'm busy dealing with something else these days. I guess the problem consists in the PCL, so I decide to install a new version .

wiedemeyer, 2017-05-23:
please use the pcl version that comes with ubuntu.

EmanuelAlvez, 2017-09-11:
@wiedemeyer  what version? i have the same problem with pcl 1.8 . I have ROS LUNAR in Ubuntu 17.04

suijingfeng, 2017-09-12:
on ubuntu 16.04,   its pcl-1.7 @EmanuelAlvez