code-iai/iai_kinect2 #403: after rosrun kinect2_viewer there isn't any response ,but starting receiver.....

ghost-knight, 2017-04-22:
rosrun kinect2_viewer kinect2_viewer kinect2 sd cloud

[INFO][main]tpoic color:/kinect2/sd/image_color_rect
[INFO][main]tpoic depth:/kinect2/sd/image_depth_rect
[INFO][main]starting receiver.....

my rosbirdge package is installed properly ,and there is not any problems about the dependences problems.
but why there is not any response

oneTop, 2017-08-29:
Hi:
    I have the same problem as you. Have you solved this problem, please?

ghost-knight, 2017-08-29:
yep i have solved it . just  make sure your kinect_bridge work properly

oneTop, 2017-08-29:
Thank you for your reply!   How should I make sure that kinect_bridge works correctly?

ubuntu@tegra-ubuntu:~$ ps -ef | grep kinect2_bridge
ubuntu    2353  2163  1 06:59 pts/2    00:00:00 /usr/bin/python /opt/ros/indigo/bin/roslaunch kinect2_bridge kinect2_bridge.launch
ubuntu    2381  2353  0 06:59 ?        00:00:00 /opt/ros/indigo/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/home/ubuntu/.ros/log/8ba78838-8c87-11e7-968b-00044b49189c/kinect2_bridge-3.log
ubuntu    2772  2734  0 06:59 pts/16   00:00:00 grep --color=auto kinect2_bridge
ubuntu@tegra-ubuntu:~$ 

The validation method above me, right?   What I'm used is "ps -ef | grep kinect_bridge"

oneTop, 2017-08-29:
[ INFO] [1503990256.546923725]: [Kinect2Bridge::main] color processing: ~75.2995ms (~13.2803Hz) publishing rate: ~13.9886Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 50.1989ms -> ~19.9207Hz
[ INFO] [1503990259.547194881]: [Kinect2Bridge::main] depth processing: ~nanms (~nanHz) publishing rate: ~0Hz
[ INFO] [1503990259.547346311]: [Kinect2Bridge::main] color processing: ~79.0003ms (~12.6582Hz) publishing rate: ~13.6649Hz
[ INFO] [1503990262.548042901]: [Kinect2Bridge::main] depth processing: ~nanms (~nanHz) publishing rate: ~0Hz
[ INFO] [1503990262.548188830]: [Kinect2Bridge::main] color processing: ~86.4357ms (~11.5693Hz) publishing rate: ~12.3296Hz
[ INFO] [1503990265.557825957]: [Kinect2Bridge::main] depth processing: ~nanms (~nanHz) publishing rate: ~0Hz
[ INFO] [1503990265.558015723]: [Kinect2Bridge::main] color processing: ~76.6023ms (~13.0544Hz) publishing rate: ~14.2872Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 50.6695ms -> ~19.7358Hz
[ INFO] [1503990268.558384334]: [Kinect2Bridge::main] depth processing: ~nanms (~nanHz) publishing rate: ~0Hz
[ INFO] [1503990268.558546181]: [Kinect2Bridge::main] color processing: ~76.3156ms (~13.1035Hz) publishing rate: ~13.997Hz

This is the part of the log that I get after I run the command: "rosrun kinect2_viewer kinect2_viewer".
It looks like there is nothing wrong with it.   Looking forward to your reply again!

remmel, 2021-12-16:
I had the same problem. Finally I restart my computer (fresh ubuntu and ros installation) and it works...
But after running in one tab `roslaunch kinect2_bridge kinect2_bridge.launch`
and in another tab `rosrun kinect2_viewer kinect2_viewer` after displaying `[ INFO] [main] starting receiver...` log, a new window `cloud viewer` should appear

TAZ010, 2022-07-15:
same problem here, is there any idea of how to solve it?