code-iai/iai_kinect2 #406: [ERROR] [registerPublisher] Failed to contact master at [localhost:11311]

Xbbei, 2017-05-03:
Hi, 
I am a newer to ROS, please forgive me my dumb question. Thank you!

When I build the iai-kinect2 successful. But I run the "kinect2_bridge", it will warning me that Failed to contact master at [localhost:11311].

rosebei@rosebei-X550JK:~/catkin_ws/devel/lib/kinect2_bridge$ gdb kinect2_bridge 
GNU gdb (Ubuntu 7.11-0ubuntu1) 7.11
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from kinect2_bridge...(no debugging symbols found)...done.
(gdb) run
Starting program: /home/rosebei/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffdb417700 (LWP 6811)]
[New Thread 0x7fffdaa04700 (LWP 6812)]
[New Thread 0x7fffda203700 (LWP 6813)]
[New Thread 0x7fffd9a02700 (LWP 6814)]
[ERROR] [registerPublisher] Failed to contact master at [localhost:11311].  Retrying...
^Z
Thread 1 "kinect2_bridge" received signal SIGTSTP, Stopped (user).
0x00007ffff5d24c5d in nanosleep () at ../sysdeps/unix/syscall-template.S:84
84	../sysdeps/unix/syscall-template.S: 没有那个文件或目录.
(gdb) 
[13]+  已停止               gdb kinect2_bridge

Thank you very much!

wiedemeyer, 2017-05-23:
http://wiki.ros.org/ROS/Tutorials

closed 2017-05-23 by wiedemeyer

kaixin-bai, 2018-11-08:
please try 'roscore' first, and then rosrun your kinect2-camera