code-iai/iai_kinect2 #409: Standalone version of kinect2_calibration (without ROS)?

floe, 2017-05-19:
Hi, I'm currently looking at a scenario where I can't use ROS, but would still like to use a separately calibrated Kinect camera. From a quick look, it seems that kinect2_calibration uses ROS only for acquiring the color/depth/IR images, is that correct? Then I'd probably give it a try to implement a standalone version that uses libfreenect2 directly.

wiedemeyer, 2017-05-23:
Hi @floe 
Yes, it just uses ROS for receiving the images. The calibration itself is done with OpenCV. I think it should be possible to reuse the code directly with libfreenect2. Another solution would be to simply calibrate the camera on a ROS system and then just copy the calibration data to the other systems.

floe, 2017-05-24:
Thanks. AFAICS kinect2_registration also isn't really tied to ROS, correct?

wiedemeyer, 2017-05-24:
kinect2_registration does not even use ROS (except for output), it is only a lib.

remmel, 2021-12-16:
Hi @floe did you managed to do something? In that case do you have a repository to share?
THanks!

floe, 2021-12-16:
Sorry, never got around to implementing this. Using Kinect Azure now.

closed 2021-12-16 by floe