code-iai/iai_kinect2 #410: OpenCV 3 support

nclslmx, 2017-05-19:
I'm working on a project using a Jetson TX2 platform which only supports Ubuntu 16.04  and as matter of fact ROS-Kinetic.  The problem is that ROS-Kinetic is build around OpenCV 3 so Kinect_Bridge doesn't work.

Do you guys know how I can work around this problem or if a Kinect_Bridge update that supports OpenCV 3 is going to be release anytime soon?

Thanks!

wiedemeyer, 2017-05-23:
kinect2_bridge should work fine with OpenCV 3. What is the error message?

ottsion, 2017-05-28:
[ INFO] [main] restoring files...
[ INFO] [CameraCalibration::readFiles] restoring file: 0000_color_points.yaml
[ INFO] [main] starting calibration...
[ INFO] [CameraCalibration::calibrateIntrinsics] calibrating intrinsics...
OpenCV Error: Assertion failed (ni == ni1) in collectCalibrationData, file /tmp/binarydeb/ros-kinetic-opencv3-3.2.0/modules/calib3d/src/calibration.cpp, line 3116
terminate called after throwing an instance of 'cv::Exception'
  what():  /tmp/binarydeb/ros-kinetic-opencv3-3.2.0/modules/calib3d/src/calibration.cpp:3116: error: (-215) ni == ni1 in function collectCalibrationData

ottsion, 2017-05-28:
`[ INFO] [main] restoring files...
[ INFO] [CameraCalibration::readFiles] restoring file: 0000_color_points.yaml
[ INFO] [main] starting calibration...
[ INFO] [CameraCalibration::calibrateIntrinsics] calibrating intrinsics...
OpenCV Error: Assertion failed (ni == ni1) in collectCalibrationData, file /tmp/binarydeb/ros-kinetic-opencv3-3.2.0/modules/calib3d/src/calibration.cpp, line 3116
terminate called after throwing an instance of 'cv::Exception'
  what():  /tmp/binarydeb/ros-kinetic-opencv3-3.2.0/modules/calib3d/src/calibration.cpp:3116: error: (-215) ni == ni1 in function collectCalibrationData
`

ottsion, 2017-05-28:
`
[ INFO] [main] restoring files...
[ INFO] [CameraCalibration::readFiles] restoring file: 0000_color_points.yaml
[ INFO] [main] starting calibration...
[ INFO] [CameraCalibration::calibrateIntrinsics] calibrating intrinsics...
OpenCV Error: Assertion failed (ni == ni1) in collectCalibrationData, file /tmp/binarydeb/ros-kinetic-opencv3-3.2.0/modules/calib3d/src/calibration.cpp, line 3116
terminate called after throwing an instance of 'cv::Exception'
  what():  /tmp/binarydeb/ros-kinetic-opencv3-3.2.0/modules/calib3d/src/calibration.cpp:3116: error: (-215) ni == ni1 in function collectCalibrationData

`

balakumar-s, 2017-06-21:
@nclslmx Did you get to fix this? I am switching over to ubuntu 16.04 + kinetic and want to know if kinect2 works.

nclslmx, 2017-06-22:
Hey! Yes it works just fine!  The problem I was having is that I wasn't publishing a tf (you probably already know that but you can modify this in the .launch) so I was not able to visualise the kinect 2 topics.  Otherwise, ever since I had this fixed , everything is working perfectly!

closed 2017-06-22 by nclslmx

balakumar-s, 2017-06-27:
Thanks. I tested and got everything working except VAAPI.