code-iai/iai_kinect2 #412: does it suppot kinetic

voyage19941119, 2017-05-22:
I'm compiling iai_kinect2 on ROS kinetic,meeting some erros .but the rgbdslamv2 requires kinetic
does it support kinetic?

wiedemeyer, 2017-05-23:
never tested it on kinetic.

marketto89, 2017-05-24:
I am using it on kinetic with no error! Can you post your errors?

SergioMP, 2017-06-12:
I'm also using it on Kinetic, in Ubuntu 16.04 x64. Doing the calibration right now.
This error:
 [ERROR] Tried to advertise a service that is already advertised in this node [/kinect2_calib_1497287035403628631/compressed/set_parameters]
is documented, but other users say that it works anyway.
What's yours?

marketto89, 2017-06-13:
Same error, but I confirm it works anyway!

aadityacr7, 2017-09-19:
Hey guys,

I am sorry I am just a beginner. I was also trying to install thison ROS-Kinetic however, in the last catkin_make, the make is failing. I know there can be multiple reasons for it 

/opt/ros/kinetic/include/opencv-3.2.0-dev/opencv2/flann/saving.h:113:63: error: exception handling disabled, use -fexceptions to enable
         throw FLANNException("Invalid index file, cannot read");

This is one of the error which is coming up. Is this because I don't have opencv-2.4 installed ??

Thanks in advance

dotdotdotdash, 2017-10-18:
Catkin_make is failing in kinetic. 

[ 90%] Linking CXX executable /home/ubuntu/jetsonbot/devel/lib/kinect2_viewer/kinect2_viewer
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmCloseOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmMap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmUnmap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevice'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevices'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevice'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmOpenOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevices'
collect2: error: ld returned 1 exit status
iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/build.make:500: recipe for target '/home/ubuntu/jetsonbot/devel/lib/kinect2_viewer/kinect2_viewer' failed
make[2]: *** [/home/ubuntu/jetsonbot/devel/lib/kinect2_viewer/kinect2_viewer] Error 1
CMakeFiles/Makefile2:2793: recipe for target 'iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/all' failed
make[1]: *** [iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

dotdotdotdash, 2017-10-18:
Never mind ! Doing this worked.

cd /usr/lib/aarch64-linux-gnu
sudo rm libGL.so
sudo ln -s /usr/lib/aarch64-linux-gnu/tegra/libGL.so libGL.so

songlin, 2018-01-05:
@aadityacr7 I'm having the same problem, any updates from you?

cshreyastech, 2019-06-03:
Have the same issue in TX2.
Making these changes didnt help. Any other suggestion please?

cd /usr/lib/aarch64-linux-gnu
sudo rm libGL.so
sudo ln -s /usr/lib/aarch64-linux-gnu/tegra/libGL.so libGL.so