code-iai/iai_kinect2 #413: when i run  "roslaunch kinect2_bridge kinect2_bridge.launch",it crashed , how can i resolve it?

ouyangandy, 2017-05-23:
[ INFO] [1495530398.216005983]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1495530398.307841897]: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/ronan/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: default
     depth_device: -1
       reg_method: default
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 5
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 14 usb devices connected
[Info] [Freenect2Impl] found 0 devices
[ERROR] [1495530398.767375377]: [Kinect2Bridge::initDevice] no Kinect2 devices found!
[ERROR] [1495530398.773057216]: [Kinect2Bridge::start] Initialization failed!
[FATAL] [1495530398.840791856]: Failed to load nodelet '/kinect2_bridge` of type `kinect2_bridge/kinect2_bridge_nodelet` to manager `kinect2'
[kinect2_bridge-3] process has died [pid 23499, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/home/ronan/.ros/log/147bc644-3f97-11e7-ab91-848f69cd9abe/kinect2_bridge-3.log].
log file: /home/ronan/.ros/log/147bc644-3f97-11e7-ab91-848f69cd9abe/kinect2_bridge-3*.log
[kinect2_bridge-3] restarting process

wiedemeyer, 2017-05-23:
Try to connect a kinect2 to you pc: `[ERROR] [1495530398.767375377]: [Kinect2Bridge::initDevice] no Kinect2 devices found!`

smallzhan, 2017-06-15:
need to install the package ros-<distro>-image-pipeline

Hontle, 2018-01-17:
@smallzhan I have installed the ros-[indigo]-image-pipeline,but it still doesnit work.
Have you solve your problem？