code-iai/iai_kinect2 #416: Obtaining SD RGB Images

stuartbourne, 2017-06-06:
Hello,

I am running the Kinect Bridge on the NVIDIA jetson, a small embedded system. I wish to stream the  RGB data from the kinect to the embedded system. The sd/image_color_rect subscribes at a rate of approximately .15 seconds, which is acceptable. However, due to the size and quality of both the qhd/image_color and hd/image_color, the subscription time of the topics is greater than 1 second for both topics. Due to this, I wish to strictly obtain the sd/image_color from the rectified sd/image_color_rect. I was curious if you had an insight on how this might happen, or any other possible workaround. Any help would be greatly appreciated.

Thanks

wuhanmoon, 2017-07-02:
Hi,
Do you find the solution? I'm using Jetson TX2.
Thanks

dkorobov2, 2017-07-31:
Same issue. Please let me know if anyone has found a solution!

zhaoxlvic, 2017-11-15:
Hi,
Do you find the solution? 
Thanks

tokyokuma, 2018-06-05:
I have same issue.
Has anybody solved this problem?

Thanks.

SriGanesh130, 2018-09-18:
I have the same problem , have anyone find the solution?

Thanks.