code-iai/iai_kinect2 #417: Difficulty in Compiling with Catkin make

KarimHassanieh, 2017-06-07:
Hi im trying to build the source code on linux with catkin make ive installed freenect and its working properly but when i use catkin make i run into this error did anyone run into something similair ? 
[image]

marketto89, 2017-06-13:
I fixed this problem by adding
`add_definitions(-fexceptions)` to the `CMakeLists.txt` of your `kinect2_registration` package.

Probably, it is due to changes in OpenCV 3.2

EmanuelAlvez, 2017-09-11:
after doing this, I still got 2 error when I made catkin_make
[image]