code-iai/iai_kinect2 #420: kinect2_bridge with "reg_method:opencl" crashes on macOS 10.12

smallzhan, 2017-06-16:
the information output is as following, while change reg_method to cpu is ok.
libfreenect2 is built with opencl and Protonect with cl argument runs smoothly

> $ rosrun kinect2_bridge kinect2_bridge                                                           
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /Users/guoqiang/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) Core(TM) i5-5257U CPU @ 2.70GHz (CPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl]   1: Intel(R) Iris(TM) Graphics 6100 (GPU)[Intel Inc.]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Intel(R) Iris(TM) Graphics 6100 (GPU)[Intel Inc.]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 4 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @20:31 with serial 007087664947
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 007087664947 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 4*128*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [Kinect2Bridge::initDevice] device serial: 007087664947
[ INFO] [Kinect2Bridge::initDevice] device firmware: 4.0.3916.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[ INFO] [*DepthRegistration::New] Using OpenCL registration method!
[ INFO] [*DepthRegistration::New] Using OpenCL registration method!
[ INFO] [DepthRegistrationOpenCL::init] devices:
[ INFO] [DepthRegistrationOpenCL::init]   0: Intel(R) Core(TM) i5-5257U CPU @ 2.70GHz
[ INFO] [DepthRegistrationOpenCL::init]   1: Intel(R) Iris(TM) Graphics 6100
[ INFO] [DepthRegistrationOpenCL::init] selected device: Intel(R) Iris(TM) Graphics 6100
[ERROR] [DepthRegistrationOpenCL::init] [depth_registration_opencl.cpp](216) data->program.build(options.c_str()) failed: -11
[ERROR] [DepthRegistrationOpenCL::init] failed to build program: -11
[ERROR] [DepthRegistrationOpenCL::init] Build Status: -2
[ERROR] [DepthRegistrationOpenCL::init] Build Options:	-D r00=9.9996023952718482e-01f -D r01=-6.1876856493501024e-03f -D r02=-6.4212079112866477e-03f -D r10=6.1580237577474372e-03f -D r11=9.9997032557939436e-01f -D r12=-4.6288987933428933e-03f -D r20=6.4496595362981421e-03f -D r21=4.5891727952671350e-03f -D r22=9.9996867020168234e-01f -D tx=-5.1309302795311242e-02f -D ty=1.9095407517132816e-04f -D tz=-7.6237637937962245e-03f -D fxR=5.3206373520231637e+02f -D fyR=5.3265461618725919e+02f -D cxR=4.7926731172994317e+02f -D cyR=2.5532884285800074e+02f -D fxRInv=1.8794740814646043e-03f -D fyRInv=1.8773891553930354e-03f -D zNear=500 -D zFar=12000 -D heightR=540 -D widthR=960 -D heightD=424 -D widthD=512
[ERROR] [DepthRegistrationOpenCL::init] Build Log:	 <program source>:190:31: error: call to '__cl_sqrt' is ambiguous
  const float4 tmp = (float4)(sqrt(2.0));
                              ^~~~~~~~~
/System/Library/Frameworks/OpenCL.framework/Versions/A/lib/clang/3.2/include/cl_kernel.h:4741:23: note: expanded from macro 'sqrt'
    #define sqrt(__x) __cl_sqrt(__x)
                      ^~~~~~~~~
/System/Library/Frameworks/OpenCL.framework/Versions/A/lib/clang/3.2/include/cl_kernel.h:4738:15: note: candidate function
__CLFN_FD_1FD(__cl_sqrt);
              ^
/System/Library/Frameworks/OpenCL.framework/Versions/A/lib/clang/3.2/include/cl_kernel.h:366:48: note: expanded from macro '__CLFN_FD_1FD'
#define __CLFN_FD_1FD(name) float __OVERLOAD__ name(float x); \
                                               ^
/System/Library/Frameworks/OpenCL.framework/Versions/A/lib/clang/3.2/include/cl_kernel.h:4738:15: note: candidate function
__CLFN_FD_1FD(__cl_sqrt);
              ^
/System/Library/Frameworks/OpenCL.framework/Versions/A/lib/clang/3.2/include/cl_kernel.h:367:21: note: expanded from macro '__CLFN_FD_1FD'
float2 __OVERLOAD__ name(float2 x); \
                    ^
/System/Library/Frameworks/OpenCL.framework/Versions/A/lib/clang/3.2/include/cl_kernel.h:4738:15: note: candidate function
__CLFN_FD_1FD(__cl_sqrt);
              ^
/System/Library/Frameworks/OpenCL.framework/Versions/A/lib/clang/3.2/include/cl_kernel.h:368:21: note: expanded from macro '__CLFN_FD_1FD'
float3 __OVERLOAD__ name(float3 x); \
                    ^
/System/Library/Frameworks/OpenCL.framework/Versions/A/lib/clang/3.2/include/cl_kernel.h:4738:15: note: candidate function
__CLFN_FD_1FD(__cl_sqrt);
              ^
/System/Library/Frameworks/OpenCL.framework/Versions/A/lib/clang/3.2/include/cl_kernel.h:369:21: note: expanded from macro '__CLFN_FD_1FD'
float4 __OVERLOAD__ name(float4 x); \
                    ^
/System/Library/Frameworks/OpenCL.framework/Versions/A/lib/clang/3.2/include/cl_kernel.h:4738:15: note: candidate function
__CLFN_FD_1FD(__cl_sqrt);
              ^
/System/Library/Frameworks/OpenCL.framework/Versions/A/lib/clang/3.2/include/cl_kernel.h:370:21: note: expanded from macro '__CLFN_FD_1FD'
float8 __OVERLOAD__ name(float8 x); \
                    ^
/System/Library/Frameworks/OpenCL.framework/Versions/A/lib/clang/3.2/include/cl_kernel.h:4738:15: note: candidate function
__CLFN_FD_1FD(__cl_sqrt);
              ^
/System/Library/Frameworks/OpenCL.framework/Versions/A/lib/clang/3.2/include/cl_kernel.h:371:22: note: expanded from macro '__CLFN_FD_1FD'
float16 __OVERLOAD__ name(float16 x);
                     ^
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[ERROR] [Kinect2Bridge::start] Initialization failed!
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing

madsherlock, 2017-10-03:
Hi,
Try changing line 190 of kinect2_registration/src/depth_registration.cl
from
` const float4 tmp = (float4)(sqrt(2.0));`
to
` const float4 tmp = (float4)(sqrt(2.0f));`

Vinzz98, 2019-10-23:
That worked for me on ubuntu 18.04. Thank you very much !