code-iai/iai_kinect2 #422: Kinect2 shows nothing from image_view

double344931987, 2017-06-20:
Hi everyone,
I try to use kinect2 to detect my object and my os is Ubuntu 14.04 indigo.
After I run roslaunch kinect2_bridge kinect2_bridge.launch and rosrun image_view image_view image:=/kinect2/qhd/image_color, I got an empty window as shown following:
screenshot from 2017-06-20 13 04 44

Then I use rostopic list and get:
/rosout
/rosout_agg
rmal@RMAL-ASUS:~/catkin_ws$ rostopic list
/kinect2/depth/camera_info
/kinect2/depth/image
/kinect2/depth/image/depth_compressed
/kinect2/depth_rect/camera_info
/kinect2/depth_rect/image
/kinect2/depth_rect/image/depth_compressed
/kinect2/depth_scaled/camera_info
/kinect2/depth_scaled/image
/kinect2/depth_scaled/image/depth_compressed
/kinect2/depth_scaled/points
/kinect2/depth_scaled_registered/camera_info
/kinect2/depth_scaled_registered/image
/kinect2/depth_scaled_registered/image/depth_compressed
/kinect2/ir/camera_info
/kinect2/ir/image
/kinect2/ir/image/compressed
/kinect2/ir_rect/camera_info
/kinect2/ir_rect/image
/kinect2/ir_rect/image/compressed
/kinect2/mono/camera_info
/kinect2/mono/image
/kinect2/mono/image/compressed
/kinect2/mono_rect/camera_info
/kinect2/mono_rect/image
/kinect2/mono_rect/image/compressed
/kinect2/mono_scaled/camera_info
/kinect2/mono_scaled/image
/kinect2/mono_scaled/image/compressed
/kinect2/rgb/camera_info
/kinect2/rgb/image
/kinect2/rgb/image/compressed
/kinect2/rgb_rect/camera_info
/kinect2/rgb_rect/image
/kinect2/rgb_rect/image/compressed
/kinect2/rgb_scaled/camera_info
/kinect2/rgb_scaled/image
/kinect2/rgb_scaled/image/compressed
/rosout
/rosout_agg

So, My window is empty due to /kinect2/hd/... /kinect2/qhd/... are not published, right?
If so, could some one give me some ideas so that I can get the image correctly?
I will be very appreciated for your help!!

suijingfeng, 2017-07-10:
my os is 16.04, I rosrun image_view image_view image:=rosrun image_view image_view image:=/kinect2/qhd/image_color_rect, and i got the image.
[image]

Abduoit, 2017-07-16:
Hi @suijingfeng 

Sorry for bothering, I need help with the following
when I run 
`roslaunch kinect2_bridge kinect2_bridge.launch`
then
`rosrun image_view image_view image:=rosrun image_view image_view image:=/kinect2/qhd/image_color_rect`

I get the following error 

```
abdulrahman@abdulrahman-ThinkPad-X230-Tablet:~$ rosrun image_view image_view image:=rosrun image_view image_view image:=/kinect2/qhd/image_color_rect
init done
[ INFO] [1500220629.085822906]: Using transport "image_view"
terminate called after throwing an instance of 'image_transport::TransportLoadException'
  what():  Unable to load plugin for transport 'image_view', error string:
According to the loaded plugin descriptions the class image_transport/image_view_sub with base class type image_transport::SubscriberPlugin does not exist. Declared types are  image_transport/compressedDepth_sub image_transport/compressed_sub image_transport/raw_sub image_transport/theora_sub
Aborted (core dumped)
```
I also tried `rosrun image_view image_view` but I get this 

```
abdulrahman@abdulrahman-ThinkPad-X230-Tablet:~$ rosrun image_view image_view
[ WARN] [1500220692.113568032]: Topic 'image' has not been remapped! Typical command-line usage:
	$ rosrun image_view image_view image:=<image topic> [transport]
init done
[ INFO] [1500220692.217649351]: Using transport "raw"
```

amaldo, 2017-07-17:
As @suijingfeng pointed out, you have to remap the image topic of the image_viewer.

closed 2017-07-17 by amaldo