code-iai/iai_kinect2 #428: could not find a package configuration file provided by "freenect2" but already have libfreenect2

ericrosenbrown, 2017-06-28:
I'm getting a similiar, but not exactly the same, problem to #231, where after the command:
catkin_make -DCMAKE_BUILD_TYPE="Release"

，there was a error:

CMake Error at iai_kinect2/kinect2_bridge/CMakeLists.txt:22 (find_package):
  Could not find a package configuration file provided by "freenect2" with
  any of the following names:

    freenect2Config.cmake
    freenect2-config.cmake

  Add the installation prefix of "freenect2" to CMAKE_PREFIX_PATH or set
  "freenect2_DIR" to a directory containing one of the above files.  If
  "freenect2" provides a separate development package or SDK, be sure it has
  been installed.

I have the path pointing to my libfreenect2 directory, but what seems to be an issue is that I can't even find either of the two cmake files anywhere in the directory. So where would I get these two cmake files to resolve the issue?

Thanks!

aadityacr7, 2017-11-07:
Hey I have this same issue.

Did you find any solution for the same ?

Thanks in advance !

buivn, 2019-03-16:
Hi,
Is there anyone solve the problem yet? Please, help