code-iai/iai_kinect2 #429: New Maintainer needed

wiedemeyer, 2017-06-29:
Hi Everyone,
since I am not working with ROS anymore and due to the lack of spare time, I am searching for people who are willing to maintain this piece of software, like making sure it works with newer ROS and libfreenect2 releases, responding to issues, etc.
So if you are using this software and you would like to use it in future ROS versions too, than think about becoming a maintainer. Just respond here. ;-)

@JimmyDaSilva wouldn't this be something for you? I saw that you frequently responded to issues to help people and did a great job with it.

bbferka, 2017-06-30:
Hi all, 

We (IAI) are willing to continue maintaining the repo, and making sure that it compiles and works for future ROS or libfreenect2 releases, mergin PRs and so on. With the departure of Thiemo, we did loose in-depth know-how about how the package works, implementation details, etc, hence we would appreciate some help from the community in responding to issues.

suijingfeng, 2017-06-30:
I also want to try.

suijingfeng, 2017-06-30:
Just give me the chance, though I‘m not very sure I have the ability.