code-iai/iai_kinect2 #432: kinect2_calibration cannot save the image

double344931987, 2017-07-06:
Hi everyone,
I am trying to use kinect2_calibration to calibrate my camera. 
I following the steps 
rosrun kinect2_bridge kinect2_bridge _fps_limit:=2
rosrun kinect2_calibration kinect2_calibration chess10x7x0.10 record color

The window shows as following
[image]

and the terminal shows as following:

[ INFO] [main] Start settings:
       Mode: record
     Source: color
      Board: chess
 Dimensions: 10 x 7
 Field size: 0.1
Dist. model: 5 coefficients
Topic color: /kinect2/hd/image_mono
   Topic ir: /kinect2/sd/image_ir
Topic depth: /kinect2/sd/image_depth
       Path: ./

[ INFO] [main] starting recorder...
[ INFO] [Recorder::startRecord] Controls:
   [ESC, q] - Exit
 [SPACE, s] - Save current frame
        [l] - decrease min and max value for IR value range
        [h] - increase min and max value for IR value range
        [1] - decrease min value for IR value range
        [2] - increase min value for IR value range
        [3] - decrease max value for IR value range
        [4] - increase max value for IR value range
[ERROR] Tried to advertise a service that is already advertised in this node [/kinect2_calib_1499375412961435695/compressed/set_parameters]
[ERROR] Tried to advertise a service that is already advertised in this node [/kinect2_calib_1499375412961435695/compressed/set_parameters]

However, whatever I press s or space, the image cannot be saved. If I press q, it can be exit. 
So, is there someone know what the problem is and any ideas are appreciate. 

Thanks in advance.

suijingfeng, 2017-07-10:
This chessboard is 6x9, not 10 * 7, you provided wrong parameters which cause cv::findChessboardCorners can not find the chessboard.

--------------------------------
      case COLOR:
        foundColor = cv::findChessboardCorners(color, boardDims, pointsColor, cv::CALIB_CB_FAST_CHECK);

-------------------------------
      if(save && ((mode == COLOR && foundColor) || (mode == IR && foundIr) || (mode == SYNC && foundColor && foundIr)))
      {
        store(color, ir, irGrey, depth, pointsColor, pointsIr);
        save = false;
      }
----------------------------------------------------------------------
findChessboardCorners will return false -> foundColor will be false -> there is no chance that 
function store will be executed.

SriGanesh130, 2018-09-19:
change 10x7 to 9x6 it will work