code-iai/iai_kinect2 #435: overlapping hand in rviz, calibrated 5 times, please help

Thompson104, 2017-07-17:
Hey all,

I tried the calibration 5 times and still I get this. I'm pretty sure this is not correct right?
[image]

Reprojection error for IR and RGB is around 0.29 and 0.124375 respectively. Synchronous is around 30.6933. I wonder if you guys have any suggestions on how to improve my result?

sitherix, 2017-07-17:
Hey what are adjustments in rviz then i can try it myself
What are you adjustments in the Grid Option?

Thompson104, 2017-07-17:
Hey,
I didn't change anything in rviz, I used ` roslaunch kinect2_bridge kinect2_bridge.launch reg_method:=cpu depth_method:=cpu publish_tf:=true
`
`rosrun rviz rviz`

Here is the grid option 
/home/drpslab/Pictures/1111.png

sitherix, 2017-07-17:
I am also new to ROS , tomorrow i will calibrate my camera, and i will see if i can help you 
What is your project about?

Thompson104, 2017-07-17:
I do this for fun, trying to show a surface in 3D coordiantes. Please let me know if you can help me with my results

amaldo, 2017-07-17:
Are you really collecting correct calibration images?  could you upload them somewhere to have a look? It almost seems like you are not exploring correctly different orientations.