code-iai/iai_kinect2 #436: Drastic drop in depth/color publishing rate

animez4me, 2017-07-17:
Hi,

After the kinect2_bridge is launched and point cloud is being displayed in rviz, the publishing rates for depth and color are about 29Hz. However after 2 minutes or so, it drastically drops to less than 5Hz and eventually less than 1Hz. Does anyone have any idea why this happens?

Thanks

amaldo, 2017-07-17:
Can you explain more about your setup?  Are you transmitting anything over WLAN?  Are both driver and rviz on the same computer? Do you see anything weird in 'dmesg'? Any errors or messages on the console where you launched the bridge?  Which ros version? Kernel version? Ubuntu version? all that info can help locate the problem.

animez4me, 2017-07-17:
Nothing over WLAN, just using the USB 3.0 to laptop. Yes just launching the bridge and rviz from terminal. I have not seen any errors or messages that look suspicious. Running ROS Indigo on Ubuntu 14.04.05, kernel 4.4.0-66-generic. Laptop specs: Intel i7, 8GB RAM, Nvidia GeForce GT 640M.
This seem to only occur when I am visualising a point or depth cloud, if I simply rostopic echo the kinect2 topics, it doesnt seem to drop. Also when the rate drops, my CPU usage spikes up to max.

animez4me, 2017-07-18:
Hi, upon checking the system log I find the messages being constantly output when the rate drops
xhci_hcd 0000:00:14.0: WARN Event TRB for slot 4 ep 8 with no TDs queued? 
I believe my laptop's Intel USB 3.0 eXtensible Host Controller should allow Kinectv2 to run?

amaldo, 2017-07-26:
Hi.

If your framerate only goes down while you are using rviz, but not while using 'rostopic echo' the problem might be that your CPU can't handle the load of rebuilding the point cloud.

When you have the problem, can you look at a cpu usage measuring tool, such as top or htop, and report which processes are using the most  CPU ?

The USB errors could also be related to too-high CPU load, as the driver does not manage to process all the data and some queues get filled up.

I have tried on a similarly spec'ed laptop (only with an intel graphics card) and don't suffer the problem you mention.