code-iai/iai_kinect2 #439: How to enable depth_registration?

TrinhTUHH, 2017-08-03:
I'm following this tutorial: http://wg-perception.github.io/tabletop/index.html#tabletop which using Object Recognition Kitchen (ORK) package to detect object. However when I run the command `rosrun rqt_reconfigure rqt_reconfigure` I don't see any camera on the GUI to enable the depth_registration as it should be. 
[image]
Do you know what is the problem?

bbferka, 2017-08-04:
@TrinhTUHH your question has nothing to do with this package (ok, maybe it does a little, but I can not know because you don't say you are using a KinectOne). Anyway, these sort of questions should be asked from the maintainers of ORK. Btw, did you actually start the launch file of any camera?. Post your question on ROS discourse. I'm closing this (non-)issue.

closed 2017-08-04 by bbferka

TrinhTUHH, 2017-08-04:
Hi, I did use this `roslaunch kinect2_bridge kinect2_bridge.launch publish_tf:=true`
and the Kinect2 works ok with rviz. I just think that there may be some problem with this lib because there is no such problem with Kinect, so I just posted it here.

bbferka, 2017-08-04:
Ah, ok, that makes more sense now :smile:  We don't offer a separate depth_registration flag for the kinect2.