code-iai/iai_kinect2 #443: How Computed depth is being calculated?

Olunia, 2017-08-13:
Hi,

In old issues I found your words:
"The depth calibration is a very preliminary piece of code. It just tries to lower the average error by subtracting it from the depth values." & "The computed one is computed using the plane the chessboard defines"
From that I understand that calibration parameters are used to calculate average error which is 1 number and then this number is being subtracted from depth data for each pixel. Am I right?

**My question here is how exactly computed depth is being calculated using calibration parameters?** 
Could you please share equations? I was not able to understand that from the code.

I also found a lot of information how to use calibration results to get X and Y coordinates, but no one mentioned how to use it to make depth data better and this is exactly what I am looking for.

I would really appreciate your help

madsherlock, 2017-10-24:
Assuming your question is specific to the "Depth calibration" parameters:
For a calibrated Kinect2, a depth calibration file is generated, called calib_depth.yaml.
This file contains one value, depthShift, in millimeters. That is the 1 number you refer to - the average error of the depth values on the chessboard (from depth image), when compared to depth computed from the known chessboard size (visible in IR image).
The author has tested that the average error (one value) is representative of ALL depth errors, and that it is independent of X and Y coordinates. You can verify this on your own dataset used for depth calibration, by inspection of the plot.dat file, which is also generated upon depth calibration. This file has a lot of data in plain text, in 5 columns: 1: X, 2: Y, 3: computed depth, 4: measured depth, 5: difference between computed and measured depth.
"Computed" means computed from known chessboard size. This is the assumed most accurate value.
"Measured" means what is in the raw depth image from Kinect2.

The "equations":
DRaw: Raw depth image, DCalib = Calibrated depth image.
For each element _i_ (pixel intensity representing millimeters) of DRaw:
DCalib(_i_)=DRaw(_i_)+_depthShift_

As you may have guessed from reading around this repository, depthShift is usually negative, and around -24 millimeters.

Please note that if you have made a depth calibration, and placed it in iai_kinect2/kinect2_bridge/data/<kinect_id>/, the calibration is applied within kinect2_bridge automatically.

Isabelp, 2017-12-08:
Hi,

I have another question regarding the depth calibration: 
according to what @madsherlock answered above, only the depth images (xx_depth.png) and the ir images (xx_ir.png) are used for the depth calibration, which made me think I can do the depth calibration without doing the sync calibration. In fact, I performed the calibration after doing color and ir calibration, skipping the step of the sync calibration, and I got the files calib_depth.yaml (-2.38e1) and plot.dat. After the tool says starting calibration ... only the xx_depth.png files were loaded. 
However, if I follow the calibration instructions (https://github.com/code-iai/iai_kinect2/tree/master/kinect2_calibration) and do color, ir and sync before the depth calibration, after starting calibration ... the tool loads xx_depth.png and xx_sync_depth.png. 

So here my question: Is my depth calibration wrong if I did not do sync calibration before?

Thanks in advance for any hint regarding this issue,

slamidiot, 2018-11-09:
@madsherlock as you said, what i understand is that,  if i want to get the right depth value, I need to subtract  "depthshift"  from the depth value of the calibrated depth image. is it right??