code-iai/iai_kinect2 #447: Could not find any suitable device

lrperlmu, 2017-09-06:
I've installed libfreenect2, and it seems to work fine on the CPU. I can run `./libfreenect2/build/bin/Protonect cpu`, and I see RGB, IR, and depth streams in the gui.

However, when I run the kinect2_bridge using `_depth_method:=cpu`, it doesn't work. I get the error `/kinect2_bridge_1504724659055964348 init:199: [DepthRegistrationOpenCL::init] could not find any suitable device`. 

Below is the full output from running `kinect2_bridge`. Please let me know if there is any more information I can provide.

Thanks!
Leah

```
$ rosrun kinect2_bridge kinect2_bridge _depth_method:=cpu 
/kinect2_bridge_1504724659055964348 initialize:289: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/leah/work/ros/kinect_cws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: cpu
     depth_device: -1
       reg_method: opencl
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
libva info: VA-API version 0.35.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
[Error] [VaapiRgbPacketProcessorImpl] vaInitialize(display, &major_ver, &minor_ver): unknown libva error
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:5 with serial 500708140242
[Info] [Freenect2Impl] found 1 devices
/kinect2_bridge_1504724659055964348 initDevice:538: [Kinect2Bridge::initDevice] Kinect2 devices found: 
/kinect2_bridge_1504724659055964348 initDevice:543: [Kinect2Bridge::initDevice]   0: 500708140242 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
/kinect2_bridge_1504724659055964348 initDevice:567: [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
/kinect2_bridge_1504724659055964348 initDevice:576: [Kinect2Bridge::initDevice] device serial: 500708140242
/kinect2_bridge_1504724659055964348 initDevice:577: [Kinect2Bridge::initDevice] device firmware: 4.0.3911.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
/kinect2_bridge_1504724659055964348 initCalibration:637: [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
/kinect2_bridge_1504724659055964348 initCalibration:642: [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
/kinect2_bridge_1504724659055964348 initCalibration:647: [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
/kinect2_bridge_1504724659055964348 initCalibration:652: [Kinect2Bridge::initCalibration] using defaults for depth shift.
/kinect2_bridge_1504724659055964348 New:81: [DepthRegistration::New] Using OpenCL registration method!
/kinect2_bridge_1504724659055964348 New:81: [DepthRegistration::New] Using OpenCL registration method!
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)
/kinect2_bridge_1504724659055964348 init:191: [DepthRegistrationOpenCL::init] devices:
/kinect2_bridge_1504724659055964348 init:199: [DepthRegistrationOpenCL::init] could not find any suitable device
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
/kinect2_bridge_1504724659055964348 start:154: [Kinect2Bridge::start] Initialization failed!
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing

```

suijingfeng, 2017-09-07:
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
[Error] [VaapiRgbPacketProcessorImpl] vaInitialize(display, &major_ver, &minor_ver): unknown libva error

     depth_method: cpu
     depth_device: -1
     reg_method: opencl

From those lines, I guess this problem is still cause by libfreenect driver, 
are you sure you install  correctly?
you must install graph card driver, OpenCL and libva-dev correctly according to you computer configuration.

Although you you specify depth processing method with _depth_method:=cpu, 
depth registration using OpenCL lib by default. they are difference.
It seem that OpenCL and libva look for Intel GPU, but you don't have one. 
Try "rosrun kinect2_bridge kinect2_bridge _depth_method:=cpu _reg_method:=cpu"?
I'm not very sure about that.
Registration is pretty cpu consuming, you can't get it on real time even through you can successful run it.

suijingfeng, 2017-09-08:
My system is ubuntu16.04 , I can run it successfully.
[image]

In my libfreenect CMakelists.txt, I set the  "Enable VA-API support" OFF:
OPTION(ENABLE_VAAPI "Enable VA-API support" OFF)

libva info: VA-API version 0.35.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
[Error] [VaapiRgbPacketProcessorImpl] vaInitialize(display, &major_ver, &minor_ver): unknown libva error

You libfreenect driver still link to VA-API.

lrperlmu, 2017-09-08:
@suijingfeng, thanks for the advice. I edited CMakelists.txt to set "Enable VA-API support" OFF (on line 37) with 
```
OPTION(ENABLE_VAAPI "Enable VA-API support" OFF)
```
Then I cleaned and recompiled both `libfreenect2` and `kinect2_bridge`. I can see that the `libva error` is no longer being printed, so that's good. However I'm still getting the same error as before: `CPU registration is not available!`

Full output pasted below.

I am running Ubuntu 14.04. I noticed that [libfreenect2](https://github.com/OpenKinect/libfreenect2#linux) provides different installation instructions for Ubuntu 14.04. Could it be that `kinect2_bridge` does not expect libfreenect2 to be installed in the way that the Ubuntu 14.04 directions specify?

Thanks!
Leah

```
$ rosrun kinect2_bridge kinect2_bridge _depth_method:=cpu _reg_method:=cpu
/kinect2_bridge_1504896883556780047 initialize:289: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/leah/work/ros/kinect_cws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: cpu
     depth_device: -1
       reg_method: cpu
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:3 with serial 500708140242
[Info] [Freenect2Impl] found 1 devices
/kinect2_bridge_1504896883556780047 initDevice:538: [Kinect2Bridge::initDevice] Kinect2 devices found: 
/kinect2_bridge_1504896883556780047 initDevice:543: [Kinect2Bridge::initDevice]   0: 500708140242 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
/kinect2_bridge_1504896883556780047 initDevice:567: [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
/kinect2_bridge_1504896883556780047 initDevice:576: [Kinect2Bridge::initDevice] device serial: 500708140242
/kinect2_bridge_1504896883556780047 initDevice:577: [Kinect2Bridge::initDevice] device firmware: 4.0.3911.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
/kinect2_bridge_1504896883556780047 initCalibration:637: [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
/kinect2_bridge_1504896883556780047 initCalibration:642: [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
/kinect2_bridge_1504896883556780047 initCalibration:647: [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
/kinect2_bridge_1504896883556780047 initCalibration:652: [Kinect2Bridge::initCalibration] using defaults for depth shift.
/kinect2_bridge_1504896883556780047 initRegistration:344: [Kinect2Bridge::initRegistration] CPU registration is not available!
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
/kinect2_bridge_1504896883556780047 start:154: [Kinect2Bridge::start] Initialization failed!
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing
```

suijingfeng, 2017-09-09:
/kinect2_bridge_1504896883556780047 initRegistration:344: [Kinect2Bridge::initRegistration] CPU registration is not available! 

Notice this line, "CPU registration is not available!" is printed by initRegistration method in kinect2_bridge.cpp.  
[image]

Dig through the source code, you will find that you cmake don‘t define conditional compilation symbols  DEPTH_REG_CPU which is caused by cmke can not find eigen3 matrix lib.
[image]
[image]
[image]

lrperlmu, 2017-09-09:
@suijingfeng this led me in the right direction! I already had eigen3 installed, but cmake could not find it because `FindEigen3.cmake` was not packaged with my version of cmake (3.2.2) or with my version of eigen3 (3.2.0-8).

```
$ cmake --find-package -DNAME=Eigen3 -DCOMPILER_ID=GNU -DLANGUAGE=C -DMODE=EXIST 
Eigen3 not found.
CMake Error: Problem processing arguments. Aborting.
$ locate FindEigen3
/usr/share/cmake-2.8/Modules/FindEigen3.cmake
$ cmake --find-package -DNAME=Eigen3 -DCOMPILER_ID=GNU -DLANGUAGE=C -DMODE=EXIST  -DCMAKE_MODULE_PATH=/usr/share/cmake-2.8/Modules
Eigen3 found.
$ cmake --version
cmake version 3.2.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```
I added the line
```
SET(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} /usr/share/cmake-2.8/Modules) 
```
in `kinect2_registration/CMakeLists.txt`, just before the line
```
find_package(Eigen3)
```
Now eigen3 is found, and I can run `kinect2_bridge` with cpu based depth registration.

Thank you!!!