code-iai/iai_kinect2 #448: Selecting camera on a multiple camera setup

pet1330, 2017-09-14:
Hi,

I have got this working using a single camera, but when I try and launch the second camera, the ros_bridge crashes. This appears to be because the second node is trying to attach to the first camera:

```bash
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 15 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:11 with serial 506428342542
[Info] [Freenect2Impl] found valid Kinect v2 @4:14 with serial 007425354147
[Info] [Freenect2Impl] found 2 devices
[ INFO] [1505394098.860618578]: [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [1505394098.860660634]: [Kinect2Bridge::initDevice]   0: 506428342542 (selected)
[ INFO] [1505394098.860693823]: [Kinect2Bridge::initDevice]   1: 007425354147
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [1505394098.955105469]: [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [1505394099.289600988]: [Kinect2Bridge::initDevice] device serial: 506428342542
[ INFO] [1505394099.289688259]: [Kinect2Bridge::initDevice] device firmware: 4.0.3916.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped

```

How do I get the package to use the second camera instead?

Thanks.

pet1330, 2017-09-14:
Never mind, just discovered the `sensor` argument

closed 2017-09-14 by pet1330

simaopp, 2018-01-05:
Can you launch two sensors on the same computer?