code-iai/iai_kinect2 #452: RGB cannot follow depth image

XuxinDoubleWang, 2017-10-13:
Dear all,
I am trying to record the point cloud using iai_kinect2 package.
Everything works fine!
However, there is still a problem.
When I move my object with a fast speed, the RGB image and depth image are separately. Specifically, the RGB image are slower than depth image, so point cloud sequence are not correct. 
If I move the object with a too slow speed, the results are correct.
Therefore, are there some methods to improve this issue so that the RGB image synchronize the depth image whatever the speed is ??

Any ideas are appreciated!!

Thanks in advance.

gpldecha, 2018-08-20:
any update on this question ? I am having the same issue.

SriGanesh130, 2018-09-25:
how to record rgb and depth frames from iai_kinect2 ?