code-iai/iai_kinect2 #453: depth packets were lost when get subscribed

mzwang828, 2017-10-22:
Hey,

Every time I subscribe to the image_depth topic, I got this depth packets were lost info:

```
[ INFO] [1508630199.610330691]: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/river/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: cpu
     depth_device: -1
       reg_method: cpu
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 5
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_39
libva error: /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so init failed
libva info: va_openDriver() returns -1
[Error] [VaapiRgbPacketProcessorImpl] vaInitialize(display, &major_ver, &minor_ver): unknown libva error
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 12 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 502777741942
[Info] [Freenect2Impl] found 1 devices
[ INFO] [1508630199.877077573]: [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [1508630199.877099390]: [Kinect2Bridge::initDevice]   0: 502777741942 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [1508630199.982712306]: [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[INFO] [1508630200.127024]: Motor controller is in CLOSED LOOP mode WITH KINEMATICS
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [1508630200.488996074]: [Kinect2Bridge::initDevice] device serial: 502777741942
[ INFO] [1508630200.489101621]: [Kinect2Bridge::initDevice] device firmware: 4.0.3911.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)
[ WARN] [1508630200.725805501]: [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [1508630200.725830402]: [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [1508630200.725842262]: [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [1508630200.725856744]: [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ INFO] [1508630200.848989642]: [DepthRegistration::New] Using CPU registration method!
[ INFO] [1508630200.849045175]: [DepthRegistration::New] Using CPU registration method!
[ INFO] [1508630201.003883002]: [Kinect2Bridge::main] waiting for clients to connect
[ INFO] [1508630220.466759786]: [Kinect2Bridge::callbackStatus] client connected. starting device...
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[Info] [DepthPacketStreamParser] 31 packets were lost
[Info] [DepthPacketStreamParser] 32 packets were lost
[ INFO] [1508630223.834978156]: [Kinect2Bridge::main] depth processing: ~17.4496ms (~57.3079Hz) publishing rate: ~2.99978Hz
[Info] [DepthPacketStreamParser] 32 packets were lost
```

Although I can still get the depth image from the topic, I just don't feel good with lost packets. When I subscribe to image_color_rect topic, I don't have any packets lost. Is there any way I can fix this packet lost problem?
Also, I tried but can't get OpenCL working on my NUC. I check beignet github and it seems they only support till 6th generation Intel, can I have openCL working with the 7th generation cpu?

I'm running on Ubuntu 16.04 with ROS kinetic. My NUC has i7-7567U  and Kabylake GT3.
I submitted another issue regarding opencl for 7th generation cpu,
https://github.com/OpenKinect/libfreenect2/issues/920

Thanks for your help!

soulslicer, 2017-11-13:
Have the same problem. It gets worse over time too. Could be LibUSB3 or hardware issues

samialperen, 2018-08-07:
It has been quite long time but this could be useful for future readers. Package losses are generally due to hardware limitations or problems while registering to libfreenect2 library. What i can suggest is to first check CPU load. If everything seems okay, then you can try to change the usb port since sometimes it could be related to LibUSB3 as previously mentioned.