code-iai/iai_kinect2 #454: kinect2_viewer error: libOpenCL.so.1: no version information available

lrxjason, 2017-11-09:
libfreenect2 worked. I can see the test window through ./Protonect
some parts of iai_kinect2 are also worked.

**But I can't run the kinect2_viewer.**

Here is the error information:
lrx@lrx-laptop:~/catkin_ws/devel/lib/kinect2_viewer$ ./kinect2_viewer 
./kinect2_viewer: /usr/local/cuda-8.0/targets/x86_64-linux/lib/libOpenCL.so.1: no version information available (required by /home/lrx/catkin_ws/devel/lib/libkinect2_registration.so)
./kinect2_viewer: /usr/local/cuda-8.0/targets/x86_64-linux/lib/libOpenCL.so.1: no version information available (required by /home/lrx/catkin_ws/devel/lib/libkinect2_registration.so)
Segmentation fault (core dumped)

ayertay, 2017-11-18:
I have the same error, after no version information it shows the following:

[ERROR] [registerPublisher] Failed to contact master at [localhost:11311].  Retrying...

cyc790, 2018-06-23:
@lrxjason I encountered the same problem,have you solved it?

ayertay, 2018-06-24:
@cyc790 I’m sorry, I gave up and switched to Kinect v1.

MorganLeee, 2019-03-04:
Try connecting Kinect2 directly with computer, not through USB-hub