code-iai/iai_kinect2 #456: image_color_rect and shape not rectangular after Calibration

astha736, 2017-11-16:
Hi, 

I recently calibrated my Kinect using Kinect2_Calibration. After the calibration, I noticed that shape of the image_color_rect is not rectangular, as it was before. I am not sure what is going on here. 

Regards,
Astha

Screenshot of image_color_rect display in rviz. 
[image]

singularTnT, 2018-05-24:
Hello, I have the same problem, any idea?
Thanks!

astha736, 2018-05-24:
Hi, 

Increasing the number of images for calibration had helped me to overcome this. Try to increase the number of images by 2x or 3x (I can't recall exactly how many images I had taken at that time ). 

I hope it helps. 

Regards,
Astha

singularTnT, 2018-05-24:
Thank you very much!
According to your solution, I think it may be that I sampled the images majorly near the center of the camera plane. I sample over 100 images for each steps of calibration, so I thought it may not be the cause of lacking data.
I will try it later to place the calibration board in the corner of the camera plane as well as the center of it.

Regards,
Liu W.