code-iai/iai_kinect2 #458: [ERROR] [DepthRegistrationOpenCL::init] [depth_registration_opencl.cpp](180) cl::Platform::get(&platforms) failed: -1001

oikk, 2017-11-25:
./Protonect works.But when i run this command,it fails
Could you please help me,thanks!
oikk@oikk-computer:~/catkin_ws$ rosrun kinect2_bridge kinect2_bridge 
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/oikk/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opengl
     depth_device: -1
       reg_method: opencl
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 8 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:5 with serial 004352764447
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 004352764447 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [Kinect2Bridge::initDevice] device serial: 004352764447
[ INFO] [Kinect2Bridge::initDevice] device firmware: 4.0.3916.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[ INFO] [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [DepthRegistration::New] Using OpenCL registration method!
[ERROR] [DepthRegistrationOpenCL::init] [depth_registration_opencl.cpp](180) cl::Platform::get(&platforms) failed: -1001
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[ERROR] [Kinect2Bridge::start] Initialization failed!
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing
oikk@oikk-computer:~/catkin_ws$

SiuKeungm, 2017-12-14:
Have you solved this problem?I also met this issue.

SiuKeungm, 2017-12-14:
maybe you can select the registration method of cpu!