code-iai/iai_kinect2 #462: Issue with asymmetric circle grid

dipendras, 2018-01-02:
I have used the asymmetric circle grid from https://nerian.com/support/resources/patterns/ (A4 size). Calibrating with this acircle is not so accurate compared to using the chessboard pattern. Any suggestions?

Note: I have used acircle4x11x0.04 while recording the images.

suijingfeng, 2018-01-02:
As far I can see, kinect2_calibration actually just a wrapper of opencv calib3d module.
Many factors can influence the calibrate accuracy, how do you evaluate the accuracy?
how many pictures you get? 
read zhang's paper before you act. 
https://www.microsoft.com/en-us/research/project/a-flexible-new-technique-for-camera-calibration-2/?from=http%3A%2F%2Fresearch.microsoft.com%2F%7Ezhang%2Fcalib%2F

what make you believe that

dipendras, 2018-01-31:
Thank you for your response. I figured out that the calibration functionality using asymmetric circle is not implemented in the repo. I have sent the merge request which contains the implementation of calibration functionality using acircle. Now it works perfectly with acircle.

Moreover with acircle, the calibration seems better even with less number of images. Cheers!

closed 2018-01-31 by dipendras