code-iai/iai_kinect2 #468: iai_kinect2 install problem

LittleFishLove, 2018-01-31:
I wanna to install iai_kinect2 on my computer. Its OS is Ubuntu 16.04. And I have installed libfreenect2 and ROS Kinetic successfully. My codes is as below:
`d ~/catkin_ws/src/`
`git clone https://github.com/code-iai/iai_kinect2.git`
`cd iai_kinect2`
`rosdep install -r --from-paths .`
`cd ~/catkin_ws`
`catkin_make -DCMAKE_BUILD_TYPE="Release"`
and in the last line, the result is:
-----------

...blabla...
[ 33%] Linking CXX shared library /home/gengruopeng/catkin_ws/devel/lib/libkinect2_registration.so
[ 33%] Built target kinect2_registration
Scanning dependencies of target kinect2_viewer
Scanning dependencies of target kinect2_bridge
Scanning dependencies of target kinect2_calibration
Scanning dependencies of target kinect2_bridge_nodelet
[ 41%] Building CXX object iai_kinect2/kinect2_calibration/CMakeFiles/kinect2_calibration.dir/src/kinect2_calibration.cpp.o
make[2]: *** No rule to make target '/usr/local/lib/libfreenect2.so', needed by '/home/gengruopeng/catkin_ws/devel/lib/libkinect2_bridge_nodelet.so'。 停止。
make[2]: *** 正在等待未完成的任务....
make[2]: *** No rule to make target '/usr/local/lib/libfreenect2.so', needed by '/home/gengruopeng/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge'。 停止。
make[2]: *** 正在等待未完成的任务....
[ 50%] Building CXX object iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o
[ 58%] Building CXX object iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/src/kinect2_bridge.cpp.o
[ 66%] Building CXX object iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/src/viewer.cpp.o
/home/gengruopeng/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:50:41: fatal error: libfreenect2/libfreenect2.hpp: 没有那个文件或目录
compilation terminated.
iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/build.make:62: recipe for target 'iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o' failed
make[2]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o] Error 1
CMakeFiles/Makefile2:1651: recipe for target 'iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/all' failed
**/home/gengruopeng/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:50:41: fatal error: libfreenect2/libfreenect2.hpp: no such file or directory
compilation terminated.**
make[1]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/build.make:62: recipe for target 'iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/src/kinect2_bridge.cpp.o' failed
make[2]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/src/kinect2_bridge.cpp.o] Error 1
CMakeFiles/Makefile2:2296: recipe for target 'iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/all' failed
make[1]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/all] Error 2
In file included from /home/gengruopeng/catkin_ws/src/iai_kinect2/kinect2_viewer/src/viewer.cpp:32:0:
/usr/include/pcl-1.7/pcl/visualization/cloud_viewer.h:202:14: warning: ‘template<class> class std::auto_ptr’ is deprecated [-Wdeprecated-declarations]
         std::auto_ptr<CloudViewer_impl> impl_;
              ^
In file included from /usr/include/c++/5/memory:81:0,
                 from /usr/include/c++/5/thread:40,
                 from /home/gengruopeng/catkin_ws/src/iai_kinect2/kinect2_viewer/src/viewer.cpp:26:
/usr/include/c++/5/bits/unique_ptr.h:49:28: note: declared here
   template<typename> class auto_ptr;
                            ^
[ 75%] Linking CXX executable /home/gengruopeng/catkin_ws/devel/lib/kinect2_calibration/kinect2_calibration
[ 75%] Built target kinect2_calibration
[ 83%] Linking CXX executable /home/gengruopeng/catkin_ws/devel/lib/kinect2_viewer/kinect2_viewer
[ 83%] Built target kinect2_viewer
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

----------
I don't know why. I successfully ran  **./bin/Protonect** ,this error made me a little pazzled.
I hope someone could help me. Thank you very much!

tokyokuma, 2018-02-05:
I'm facing same problem....
Have you solved it already?

mohamedsemz, 2018-02-06:
i have the same problem any solution ?

LittleFishLove, 2018-02-08:
I have resolve this problem, I reinstall Ubuntu and take steps as tutorials, and the problem resolved. And I must mention, there may have been some problems before this problem, so you'd better reinstall libfreenect2 or  ROS ,and make sure no problems happen, maybe this problem won't occur again!

closed 2018-02-08 by LittleFishLove