code-iai/iai_kinect2 #469: Mismatch of HD and SD while calibrating two sensors extrinsically

jdeeee, 2018-02-01:
Hey,

I am experiencing the following situation, which I can't really explain to myself:
When I use a checkerboard to calibrate two Kinect v2 extrinsically using the HD-RGB images, the result will be valid for the HD-pointClouds of both cameras (i.e., they align and the world origin is in the correct checkerboard location). When I use this extrinsic calibration result for SD-pointClouds while accounting for the transformation between cameraX_ir_optical_frame and cameraX_rgb_optical frame, the result is not good. There seems to be a significant offset, e.g., the world origin is actually a couple of centimeters away from the correct checkerboard location (in both cameras point clouds).
When I do the above using the IR-images or SD-RGB-images for extrinsic calibration of the two sensors, the result is valid for my SD-pointClouds.

Why would that be the case? One might think that it has to do with a bad calibration achieved by kinect2_calibration. However, the color is aligned nicely on all respective point clouds and when I plot a sd and hd point cloud of the same sensor in the same frame I can't observe the mentioned offset. 

I thought this issue might be related to #160 and #173 even though I believe I do not have the problem of misaligned sd and hd point clouds, if I understood their problem correctly.

JimmyDaSilva, 2018-02-01:
Hi @jdeeee for coming out on this issue as well. We have exactly the same problem here.

@wiedemeyer  Let's put my extrinsics calibration to the robot aside. Thinking more about this I came out with a much more self explanatory demonstration of the problem.Anyone can try this, because it doesn't require a robot.

Just display the HD and SD pointclouds at the same time with different colors on RVIZ.
[image]
There is definitely a shift between the two. And it's between 5 and 10cm

jdeeee, 2018-02-01:
Hi!

I thought so too, but I am not so sure anymore. Because, as I stated, I can't observe the mentioned offset when I plot the hd and sd cloud in the same frame (maybe its a bit difficult to see on the picture): 
[image]
@JimmyDaSilva: QHD and HD should both be in the same frame. Do you experience the same issue, when choosing "kinectV2_rgb_optical_frame" as fixed frame under global options?

JimmyDaSilva, 2018-02-01:
@jdeeee So we have different problems then.
Same with HD or QHD. Changing fixed_frame on RVIZ won't change anything.

@wiedemeyer I still don't understand why I observe this shift. I remember recalibrating at least 3 times. Did you make any important changes in the calibration process in the last year ?
I can't remember, do you use the factory internal offsets or the ones calculated during the kinect2_calibration for the distance between ir and rgb ?

Thanks ;)

jdeeee, 2018-02-02:
Hey,
I have made some big changes to my original post, since some of my observations were simply wrong. @JimmyDaSilva we seem to have different problems here, but maybe #173 is still related...

@wiedemeyer Are you still working on this package and do you have an idea what is going on?

SiddhantNadkarni, 2020-07-31:
Having same issue. Errors in calibration when I use HD images from Kinect itself, or even an external camera, there is a shift in world origin. Please let me know if anyone has managed to resolve this issue. Thanks