code-iai/iai_kinect2 #474: Failed to load nodelet '/kinect2_points_xyzrgb_hd`

QinZiwen, 2018-02-07:
When I compile, there is a warning
```
[ 92%] Building CXX object turtlebot_exploration_3d/CMakeFiles/turtlebot_exploration_3d.dir/src/gpregressor.cpp.o
/usr/bin/ld: warning: libopencv_imgproc.so.3.2, needed by /home/turtlebot/catkin_ws/devel/lib/libkinect2_registration.so, may conflict with libopencv_imgproc.so.2.4
[ 92%] Built target kinect2_bridge
[100%] Building CXX object turtlebot_exploration_3d/CMakeFiles/turtlebot_exploration_3d.dir/src/covMaterniso3.cpp.o
[100%] Built target kinect2_calibration
Linking CXX executable /home/turtlebot/catkin_ws/devel/lib/kinect2_viewer/kinect2_viewer
/usr/bin/ld: warning: libopencv_calib3d.so.2.4, needed by /usr/lib/x86_64-linux-gnu/libopencv_contrib.so.2.4.8, may conflict with libopencv_calib3d.so.3.2
[100%] Built target kinect2_viewer
```

When run “roslaunch kinect2_bridge kinect2_bridge.launch”,there are some mistakes,of which the main one is "Failed to load nodelet '/kinect2_points_xyzrgb_hd` of type depth_image_proc/point_cloud_xyzrgb` to manager `kinect2'"

My log:
```
$ roslaunch kinect2_bridge kinect2_bridge.launch
... logging to /home/turtlebot/.ros/log/a5e1abee-0c01-11e8-8592-c8d3ffd835e0/roslaunch-turtlebot-OMEN-by-HP-Laptop-13973.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:53264/

SUMMARY
========

PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/turtlebot/c...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 12.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: False
 * /kinect2_bridge/queue_size: 5
 * /kinect2_bridge/reg_device: -1
 * /kinect2_bridge/reg_method: default
 * /kinect2_bridge/sensor: 
 * /kinect2_bridge/use_png: False
 * /kinect2_bridge/worker_threads: 4
 * /kinect2_points_xyzrgb_hd/queue_size: 5
 * /kinect2_points_xyzrgb_qhd/queue_size: 5
 * /kinect2_points_xyzrgb_sd/queue_size: 5
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /
    kinect2 (nodelet/nodelet)
    kinect2_bridge (nodelet/nodelet)
    kinect2_points_xyzrgb_hd (nodelet/nodelet)
    kinect2_points_xyzrgb_qhd (nodelet/nodelet)
    kinect2_points_xyzrgb_sd (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [13985]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a5e1abee-0c01-11e8-8592-c8d3ffd835e0
process[rosout-1]: started with pid [13998]
started core service [/rosout]
process[kinect2-2]: started with pid [14008]
process[kinect2_bridge-3]: started with pid [14016]
[... 62 lines cut ...]
log file: /home/turtlebot/.ros/log/a5e1abee-0c01-11e8-8592-c8d3ffd835e0/kinect2_points_xyzrgb_sd-4*.log
[kinect2_bridge-3] restarting process
process[kinect2_bridge-3]: started with pid [14169]
[kinect2_points_xyzrgb_sd-4] restarting process
process[kinect2_points_xyzrgb_sd-4]: started with pid [14170]
[ INFO] [1518006187.434549214]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1518006187.437629394]: waitForService: Service [/kinect2/load_nodelet] could not connect to host [localhost:60674], waiting...
[kinect2_points_xyzrgb_qhd-5] process has died [pid 14018, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/qhd/camera_info rgb/image_rect_color:=kinect2/qhd/image_color_rect depth_registered/image_rect:=kinect2/qhd/image_depth_rect depth_registered/points:=kinect2/qhd/points __name:=kinect2_points_xyzrgb_qhd __log:=/home/turtlebot/.ros/log/a5e1abee-0c01-11e8-8592-c8d3ffd835e0/kinect2_points_xyzrgb_qhd-5.log].
log file: /home/turtlebot/.ros/log/a5e1abee-0c01-11e8-8592-c8d3ffd835e0/kinect2_points_xyzrgb_qhd-5*.log
[kinect2_points_xyzrgb_hd-6] process has died [pid 14028, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/hd/camera_info rgb/image_rect_color:=kinect2/hd/image_color_rect depth_registered/image_rect:=kinect2/hd/image_depth_rect depth_registered/points:=kinect2/hd/points __name:=kinect2_points_xyzrgb_hd __log:=/home/turtlebot/.ros/log/a5e1abee-0c01-11e8-8592-c8d3ffd835e0/kinect2_points_xyzrgb_hd-6.log].
log file: /home/turtlebot/.ros/log/a5e1abee-0c01-11e8-8592-c8d3ffd835e0/kinect2_points_xyzrgb_hd-6*.log
[kinect2_points_xyzrgb_qhd-5] restarting process
process[kinect2_points_xyzrgb_qhd-5]: started with pid [14203]
[kinect2_points_xyzrgb_hd-6] restarting process
process[kinect2_points_xyzrgb_hd-6]: started with pid [14204]
^C[kinect2_points_xyzrgb_hd-6] killing on exit
[kinect2_points_xyzrgb_qhd-5] killing on exit
[kinect2_points_xyzrgb_sd-4] killing on exit
[kinect2_bridge-3] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

QinZiwen, 2018-02-07:
When I install opencv2.4.13, the problem is solved

Jackey-Huo, 2019-05-29:
Hi is there any progress in the issue? I meet the same problem

florianspy, 2022-04-01:
I found out you can even use opencv 3.2 but only when cv bridge was also build with opencv 3.2 as it is for ros melodic. So if you have this issue above the opencv version used for building kinect bridge and building cv_bridge of ros do not match. So make them match and this will solve the above issue.