code-iai/iai_kinect2 #475: kinect2_bridge.launch issue

csyslcp, 2018-02-13:
I got my kinect2 working previously but then it seems that I broke my system. And when I roslaunch kinect2_bridge kinect2_bridge.launch. It meet some fatal problems.

: 

... logging to /home/dgjks/.ros/log/731dc518-10ce-11e8-8ed1-a0c5894e2889/roslaunch-dgjks-26210.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://dgjks:45657/

SUMMARY
========

PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/dgjks/catki...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 12.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: False
 * /kinect2_bridge/queue_size: 5
 * /kinect2_bridge/reg_device: -1
 * /kinect2_bridge/reg_method: default
 * /kinect2_bridge/sensor: 
 * /kinect2_bridge/use_png: False
 * /kinect2_bridge/worker_threads: 4
 * /kinect2_points_xyzrgb_hd/queue_size: 5
 * /kinect2_points_xyzrgb_qhd/queue_size: 5
 * /kinect2_points_xyzrgb_sd/queue_size: 5
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /
    kinect2 (nodelet/nodelet)
    kinect2_bridge (nodelet/nodelet)
    kinect2_points_xyzrgb_hd (nodelet/nodelet)
    kinect2_points_xyzrgb_qhd (nodelet/nodelet)
    kinect2_points_xyzrgb_sd (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [26222]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 731dc518-10ce-11e8-8ed1-a0c5894e2889
process[rosout-1]: started with pid [26235]
started core service [/rosout]
process[kinect2-2]: started with pid [26246]
process[kinect2_bridge-3]: started with pid [26253]
process[kinect2_points_xyzrgb_sd-4]: started with pid [26254]
process[kinect2_points_xyzrgb_qhd-5]: started with pid [26256]
[ INFO] [1518533951.946963780]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1518533951.949422291]: waitForService: Service [/kinect2/load_nodelet] has not been advertised, waiting...
process[kinect2_points_xyzrgb_hd-6]: started with pid [26269]
[ INFO] [1518533951.969126134]: Initializing nodelet with 8 worker threads.
[ INFO] [1518533951.970471302]: waitForService: Service [/kinect2/load_nodelet] is now available.
[ INFO] [1518533951.999968828]: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/dgjks/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: default
     depth_device: -1
       reg_method: default
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 5
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) HD Graphics Skylake Halo GT4 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Intel(R) HD Graphics Skylake Halo GT4 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 8 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 007988365247
[Info] [Freenect2Impl] found 1 devices
[ INFO] [1518533952.627013851]: [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [1518533952.627057084]: [Kinect2Bridge::initDevice]   0: 007988365247 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [1518533952.721266156]: [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [1518533953.084428506]: [Kinect2Bridge::initDevice] device serial: 007988365247
[ INFO] [1518533953.084535736]: [Kinect2Bridge::initDevice] device firmware: 4.0.3916.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[FATAL] [1518533953.436930062]: Failed to load nodelet '/kinect2_points_xyzrgb_hd` of type `depth_image_proc/point_cloud_xyzrgb` to manager `kinect2'
[FATAL] [1518533953.436972423]: Failed to load nodelet '/kinect2_bridge` of type `kinect2_bridge/kinect2_bridge_nodelet` to manager `kinect2'
[FATAL] [1518533953.436975238]: Failed to load nodelet '/kinect2_points_xyzrgb_qhd` of type `depth_image_proc/point_cloud_xyzrgb` to manager `kinect2'
[FATAL] [1518533953.437066215]: Failed to load nodelet '/kinect2_points_xyzrgb_sd` of type `depth_image_proc/point_cloud_xyzrgb` to manager `kinect2'
[kinect2-2] process has died [pid 26246, exit code -11, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=kinect2 __log:=/home/dgjks/.ros/log/731dc518-10ce-11e8-8ed1-a0c5894e2889/kinect2-2.log].
log file: /home/dgjks/.ros/log/731dc518-10ce-11e8-8ed1-a0c5894e2889/kinect2-2*.log
[kinect2_bridge-3] process has died [pid 26253, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/home/dgjks/.ros/log/731dc518-10ce-11e8-8ed1-a0c5894e2889/kinect2_bridge-3.log].
log file: /home/dgjks/.ros/log/731dc518-10ce-11e8-8ed1-a0c5894e2889/kinect2_bridge-3*.log
[kinect2_bridge-3] restarting process
process[kinect2_bridge-3]: started with pid [26407]
[ INFO] [1518533953.677877104]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1518533953.680948464]: waitForService: Service [/kinect2/load_nodelet] could not connect to host [dgjks:47998], waiting...
[kinect2_points_xyzrgb_sd-4] process has died [pid 26254, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/sd/camera_info rgb/image_rect_color:=kinect2/sd/image_color_rect depth_registered/image_rect:=kinect2/sd/image_depth_rect depth_registered/points:=kinect2/sd/points __name:=kinect2_points_xyzrgb_sd __log:=/home/dgjks/.ros/log/731dc518-10ce-11e8-8ed1-a0c5894e2889/kinect2_points_xyzrgb_sd-4.log].
log file: /home/dgjks/.ros/log/731dc518-10ce-11e8-8ed1-a0c5894e2889/kinect2_points_xyzrgb_sd-4*.log
[kinect2_points_xyzrgb_qhd-5] process has died [pid 26256, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/qhd/camera_info rgb/image_rect_color:=kinect2/qhd/image_color_rect depth_registered/image_rect:=kinect2/qhd/image_depth_rect depth_registered/points:=kinect2/qhd/points __name:=kinect2_points_xyzrgb_qhd __log:=/home/dgjks/.ros/log/731dc518-10ce-11e8-8ed1-a0c5894e2889/kinect2_points_xyzrgb_qhd-5.log].
log file: /home/dgjks/.ros/log/731dc518-10ce-11e8-8ed1-a0c5894e2889/kinect2_points_xyzrgb_qhd-5*.log
[kinect2_points_xyzrgb_hd-6] process has died [pid 26269, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/hd/camera_info rgb/image_rect_color:=kinect2/hd/image_color_rect depth_registered/image_rect:=kinect2/hd/image_depth_rect depth_registered/points:=kinect2/hd/points __name:=kinect2_points_xyzrgb_hd __log:=/home/dgjks/.ros/log/731dc518-10ce-11e8-8ed1-a0c5894e2889/kinect2_points_xyzrgb_hd-6.log].
log file: /home/dgjks/.ros/log/731dc518-10ce-11e8-8ed1-a0c5894e2889/kinect2_points_xyzrgb_hd-6*.log
[kinect2_points_xyzrgb_sd-4] restarting process
process[kinect2_points_xyzrgb_sd-4]: started with pid [26423]
[kinect2_points_xyzrgb_qhd-5] restarting process
process[kinect2_points_xyzrgb_qhd-5]: started with pid [26424]
[kinect2_points_xyzrgb_hd-6] restarting process
process[kinect2_points_xyzrgb_hd-6]: started with pid [26426]

What should I do... Thanks for you help!

pholthau, 2018-02-22:
Same here, anyone has an idea what happens? I followed the installation instructions on a fresh system.

pholthau, 2018-02-23:
Okay, I got it working by connecting to the computer via ssh *without* using X forwarding.

15061885704, 2018-03-24:
@pholthau hello,can you tell me how did you make it about "via ssh without using X forwarding"?I am a new here in this area and I would appreciate it if you can help me, thank you.

pholthau, 2018-03-26:
@15061885704 sure. I accessed the computer with the Kinect (called pc2) remotely from a different machine. In my first attempt, i typed in ``ssh -X pc2``. When using ``-X`` all commands that open a new window (X window) will be displayed on the machine from which you are connecting. This is called X forwarding. If you do *not* use ``-X`` but type in ``ssh pc2`` only, windows are not displayed at all (and the program closes with an error) or they are displayed on the remote machine's graphical output.

In my case, using only ``ssh pc2`` worked fine. I assume that X forwarding doesn't mix very well with OpenGL or some other part of the image pipeline. If you do not have a setup where you connect remotely to the PC with the Kinect attached, this should not affect you.

15061885704, 2018-03-27:
@pholthau well,I got it and now it works fine with me,thank you!

pholthau, 2018-03-27:
Good to hear!