code-iai/iai_kinect2 #482: Compile Error about OpenCV multi-version: a lot of "undefined reference"

XunshanMan, 2018-04-06:
Hello! Thanks for visiting.
I have been stuck in the issue for several days. 
I have successfully installed libfreenect2 and run ./Prototect and seen the output of rgb and depth images.

My computer has OpenCV 2.4.8( installed with ROS indigo), and OpenCV 3.2 compiled from source code with contrib modules. At first I compiled successfully but showed "segmentation fault" and crashed when I ran kinect2_bridge, and then I found out that I was compiled with OpenCV 3.2. So I changed it into OpenCV 2.4.8, but it got a lot of "undefined reference" error showing below.

I could make sure that I was using OpenCV 2.8 because in the CMakeLIsts.txt, I output ${OpenCV_LIBRARIES} and ${OpenCV_INCLUDE_DIRS}, which showed the path of version 2.4.8.

In my computer, I checked the paths of two opencv versions:
> For OpenCV 3.2
include directories:
/usr/local/include
/usr/local/include/opencv

libs: /usr/local/lib

> For OpenCV 2.4.8
include directories:
/usr/include/
/usr/include/opencv

libs: /usr/lib/x86_64-linux-gnu

Thanks so much for the help! 
Output is as below.

XunshanMan, 2018-04-06:
Here is the output of "catkin_make":
`Base path: /home/jk/catkin_lzw
Source space: /home/jk/catkin_lzw/src
Build space: /home/jk/catkin_lzw/build
Devel space: /home/jk/catkin_lzw/devel
Install space: /home/jk/catkin_lzw/install
####
#### Running command: "make cmake_check_build_system" in "/home/jk/catkin_lzw/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/jk/catkin_lzw/devel
-- Using CMAKE_PREFIX_PATH: /home/jk/catkin_lzw/devel;/home/jk/catkin_ros_qi/devel;/opt/ros/indigo
-- This workspace overlays: /home/jk/catkin_lzw/devel;/home/jk/catkin_ros_qi/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/jk/catkin_lzw/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.19
-- BUILD_SHARED_LIBS is on
-- Using CATKIN_WHITELIST_PACKAGES: kinect2_bridge
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~ traversing 1 packages in topological order:
-- ~~ - kinect2_bridge
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'kinect2_bridge'
-- ==> add_subdirectory(iai_kinect2/kinect2_bridge)
-- Using these message generators: gencpp;genlisp;genpy
**---->OpenCV status:
---->OpenCV_INCLUDE_DIRS ---- 
/usr/include/opencv/usr/include
---->OpenCV_LIBRARIES ---- 
opencv_videostabopencv_videoopencv_superresopencv_stitchingopencv_photoopencv_oclopencv_objdetectopencv_mlopencv_legacyopencv_imgprocopencv_highguiopencv_gpuopencv_flannopencv_features2dopencv_coreopencv_contribopencv_calib3d**
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jk/catkin_lzw/build
####
#### Running command: "make -j4 -l4" in "/home/jk/catkin_lzw/build"
####
Linking CXX executable /home/jk/catkin_lzw/devel/lib/kinect2_bridge/kinect2_bridge
[ 50%] Built target kinect2_bridge_nodelet
CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o: In function `cvflann::anyimpl::big_any_policy<cv::String>::static_delete(void**)':
kinect2_bridge.cpp:(.text._ZN7cvflann7anyimpl14big_any_policyIN2cv6StringEE13static_deleteEPPv[_ZN7cvflann7anyimpl14big_any_policyIN2cv6StringEE13static_deleteEPPv]+0x15): undefined reference to `cv::String::deallocate()'
CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o: In function `cvflann::anyimpl::big_any_policy<cv::String>::move(void* const*, void**)':
kinect2_bridge.cpp:(.text._ZN7cvflann7anyimpl14big_any_policyIN2cv6StringEE4moveEPKPvPS5_[_ZN7cvflann7anyimpl14big_any_policyIN2cv6StringEE4moveEPKPvPS5_]+0x10): undefined reference to `cv::String::deallocate()'
kinect2_bridge.cpp:(.text._ZN7cvflann7anyimpl14big_any_policyIN2cv6StringEE4moveEPKPvPS5_[_ZN7cvflann7anyimpl14big_any_policyIN2cv6StringEE4moveEPKPvPS5_]+0x24): undefined reference to `cv::String::deallocate()'
CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o: In function `cv::operator<<(std::ostream&, cv::Mat const&)':
kinect2_bridge.cpp:(.text.unlikely+0x27): undefined reference to `cv::Formatter::get(int)'
CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o: In function `cv::String::String(char const*)':
kinect2_bridge.cpp:(.text._ZN2cv6StringC2EPKc[_ZN2cv6StringC5EPKc]+0x30): undefined reference to `cv::String::allocate(unsigned long)'
CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o: In function `cv::Mat::Mat(int, int, int, void*, unsigned long)':
kinect2_bridge.cpp:(.text._ZN2cv3MatC2EiiiPvm[_ZN2cv3MatC5EiiiPvm]+0x16d): undefined reference to `cv::String::allocate(unsigned long)'
kinect2_bridge.cpp:(.text._ZN2cv3MatC2EiiiPvm[_ZN2cv3MatC5EiiiPvm]+0x1c7): undefined reference to `cv::error(int, cv::String const&, char const*, char const*, int)'
kinect2_bridge.cpp:(.text._ZN2cv3MatC2EiiiPvm[_ZN2cv3MatC5EiiiPvm]+0x1cf): undefined reference to `cv::String::deallocate()'
kinect2_bridge.cpp:(.text._ZN2cv3MatC2EiiiPvm[_ZN2cv3MatC5EiiiPvm]+0x1fd): undefined reference to `cv::String::allocate(unsigned long)'
kinect2_bridge.cpp:(.text._ZN2cv3MatC2EiiiPvm[_ZN2cv3MatC5EiiiPvm]+0x24a): undefined reference to `cv::error(int, cv::String const&, char const*, char const*, int)'
kinect2_bridge.cpp:(.text._ZN2cv3MatC2EiiiPvm[_ZN2cv3MatC5EiiiPvm]+0x252): undefined reference to `cv::String::deallocate()'
kinect2_bridge.cpp:(.text._ZN2cv3MatC2EiiiPvm[_ZN2cv3MatC5EiiiPvm]+0x269): undefined reference to `cv::String::deallocate()'
kinect2_bridge.cpp:(.text._ZN2cv3MatC2EiiiPvm[_ZN2cv3MatC5EiiiPvm]+0x27c): undefined reference to `cv::String::deallocate()'
CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o: In function `Kinect2Bridge::createCompressed(cv::Mat const&, std_msgs::Header_<std::allocator<void> > const&, Kinect2Bridge::Image, sensor_msgs::CompressedImage_<std::allocator<void> >&) const':
kinect2_bridge.cpp:(.text._ZNK13Kinect2Bridge16createCompressedERKN2cv3MatERKN8std_msgs7Header_ISaIvEEENS_5ImageERN11sensor_msgs16CompressedImage_IS6_EE[_ZNK13Kinect2Bridge16createCompressedERKN2cv3MatERKN8std_msgs7Header_ISaIvEEENS_5ImageERN11sensor_msgs16CompressedImage_IS6_EE]+0xe3): undefined reference to `cv::String::allocate(unsigned long)'
kinect2_bridge.cpp:(.text._ZNK13Kinect2Bridge16createCompressedERKN2cv3MatERKN8std_msgs7Header_ISaIvEEENS_5ImageERN11sensor_msgs16CompressedImage_IS6_EE[_ZNK13Kinect2Bridge16createCompressedERKN2cv3MatERKN8std_msgs7Header_ISaIvEEENS_5ImageERN11sensor_msgs16CompressedImage_IS6_EE]+0x [... cut ...]
kinect2_bridge.cpp:(.text._ZNK13Kinect2Bridge24loadCalibrationDepthFileERKSsRd[_ZNK13Kinect2Bridge24loadCalibrationDepthFileERKSsRd]+0x2b2): undefined reference to `cv::String::deallocate()'
kinect2_bridge.cpp:(.text._ZNK13Kinect2Bridge24loadCalibrationDepthFileERKSsRd[_ZNK13Kinect2Bridge24loadCalibrationDepthFileERKSsRd]+0x2c6): undefined reference to `cv::String::deallocate()'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/jk/catkin_lzw/devel/lib/kinect2_bridge/kinect2_bridge] Error 1
make[1]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed`

fengjiufa, 2019-11-30:
Hello, Do you sovle it? I faced it samely, i hope you help me .
Thanks!

XunshanMan, 2019-11-30:
> Hello, Do you sovle it? I faced it samely, i hope you help me .
> Thanks!

Hi, since it have been a long time, I forgot the detail.
But I'm sure the problem "undefined reference to.." happens because of linking to the wrong libs of OpenCV. For example, you include .h files of opencv3, but you link to libs of opencv2.
If you install only one OpenCV version on your PC, I think the problem will not occur.