code-iai/iai_kinect2 #487: problem in depth calibration,when i press spacebar，there is no ir image that are saved，please help

TouchDeeper, 2018-05-15:
i run "rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 record ir" to catch ir image, when i press spacebar，there is no ir image that are saved，but when i run "rostopic echo" /kinect2/hd/image_depth_rect",there have depth data,what's problem? please help.

TouchDeeper, 2018-12-19:
Oh,i have solved the problem, but i forget to close the issue,sorry .Don't use the ink printing to print the calibration board. When i use the laser printing, the problem solved

closed 2018-12-19 by TouchDeeper

Neur1n, 2019-03-20:
I have the same issue, but what do you mean, the printed chess board has an effect on the code?

TouchDeeper, 2019-03-20:
Yes. Don't use the ink printing to print the calibration board. Just use laser printing.

Neur1n, 2019-03-20:
I understand now, the detection depends on the printing. But in my case, I have a `6x9` chess board instead of `7x10` (see #432). Thanks for the response any way. :)