code-iai/iai_kinect2 #490: Tried to advertise a service that is already advertised in this node

QinZiwen, 2018-05-30:
I want to calibrate kinect2 follow https://github.com/code-iai/iai_kinect2/tree/master/kinect2_calibration
but when I run 
```
$ rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 record color
```
Terminal show error for me:
```
$ rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 record color
[ INFO] [main] Start settings:
       Mode: record
     Source: color
      Board: chess
 Dimensions: 5 x 7
 Field size: 0.03
Dist. model: 5 coefficients
Topic color: /kinect2/hd/image_mono
   Topic ir: /kinect2/sd/image_ir
Topic depth: /kinect2/sd/image_depth
       Path: ./

[ INFO] [main] starting recorder...
[ INFO] [Recorder::startRecord] Controls:
   [ESC, q] - Exit
 [SPACE, s] - Save current frame
        [l] - decrease min and max value for IR value range
        [h] - increase min and max value for IR value range
        [1] - decrease min value for IR value range
        [2] - increase min value for IR value range
        [3] - decrease max value for IR value range
        [4] - increase max value for IR value range
[ERROR] Tried to advertise a service that is already advertised in this node [/kinect2_calib_1527671799513392556/compressed/set_parameters]
[ERROR] Tried to advertise a service that is already advertised in this node [/kinect2_calib_1527671799513392556/compressed/set_parameters]
```
And I do not save image by push space or 's'

Thanks

buckmole, 2018-07-20:
I have met the same situation, don't know how to deal with it yet.

ericrosenbrown, 2018-07-29:
@buckmole I am also having this issue, have you found a workaround for this yet?

heyonghong, 2018-12-17:
I am also having this problem, have you solved it?

dhiegomaga, 2019-02-05:
Same issue...

evalliETH, 2019-06-25:
Where should the viewer image be? When i type`rosrun kinect2_calibration kinect2_calibration record color chess4x3x0.10 -color /kinect2_head/mono/image -ir /kinect2_head/ir/image -interval 1 ~/tmp/` and the error occurs, nothing else appears?