code-iai/iai_kinect2 #496: catkin_make error, Invoking failed, undefined reference to ros and pcl

xuemei-ye, 2018-07-03:
I  follow the instructions of iai_kinect, and libfreenect can worl correctly, but when I run catkin_make -DCMAKE_BUILD_TYPE="Release", it failed. I look almost all the related issues , and tried these solutions. The mistake has been going on for several days,  Is there a solution for this error ? Am I miss something ?  Thank you !

`myubuntu@myubuntu:~/catkin_ws$ catkin_make -DCMAKE_BUILD_TYPE="Release"
Base path: /home/myubuntu/catkin_ws
Source space: /home/myubuntu/catkin_ws/src
Build space: /home/myubuntu/catkin_ws/build
Devel space: /home/myubuntu/catkin_ws/devel
Install space: /home/myubuntu/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/myubuntu/catkin_ws/build"
####
####
#### Running command: "make -j12 -l12" in "/home/myubuntu/catkin_ws/build"
####
[ 27%] Built target kinect2_registration
[ 36%] Linking CXX executable /home/myubuntu/catkin_ws/devel/lib/kinect2_calibration/kinect2_calibration
[ 45%] Linking CXX executable /home/myubuntu/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge
[ 63%] Built target kinect2_bridge_nodelet
[ 72%] Linking CXX executable /home/myubuntu/catkin_ws/devel/lib/kinect2_viewer/kinect2_viewer
CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o: In function `Kinect2Bridge::main()':
kinect2_bridge.cpp:(.text.startup._ZN13Kinect2Bridge4mainEv[_ZN13Kinect2Bridge4mainEv]+0x669): undefined reference to `ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > const&, char const*, int, char const*)'
kinect2_bridge.cpp:(.text._ZN13Kinect2Bridge15initCalibrationERKSsS1_[_ZN13Kinect2Bridge15initCalibrationERKSsS1_]+0x28e7): undefined reference to `ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > const&, char const*, int, char const*)'
CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o: In function `Kinect2Bridge::initialize()':
kinect2_bridge.cpp:(.text._ZN13Kinect2Bridge10initializeEv[_ZN13Kinect2Bridge10initializeEv]+0xe5): undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'
kinect2_bridge.cpp:(.text._ZN13Kinect2Bridge10initializeEv[_ZN13Kinect2Bridge10initializeEv]+0x102): undefined reference to `ros::NodeHandle::getParam(std::string const&, std::string&) const'
kinect2_bridge.cpp:(.text._ZN13Kinect2Bridge10initializeEv[_ZN13Kinect2Bridge10initializeEv]+0x184): undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'
kinect2_bridge.cpp:(.text._ZN13Kinect2Bridge10initializeEv[_ZN13Kinect2Bridge10initializeEv]+0x1a8): undefined reference to `ros::NodeHandle::getParam(std::string const&, std::string&) const'
kinect2_bridge.cpp:(.text._ZN13Kinect2Bridge10initializeEv[_ZN13Kinect2Bridge10initializeEv]+0x216): undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'
kinect2_bridge.cpp:(.text._ZN13Kinect2Bridge10initializeEv[_ZN13Kinect2Bridge10initializeEv]+0x285): undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'
kinect2_bridge.cpp:(.text._ZN13Kinect2Bridge10initializeEv[_ZN13Kinect2Bridge10initializeEv]+0x2a2): undefined reference to `ros::NodeHandle::getParam(std::string const&, std::string&) const'
kinect2_bridge.cpp:(.text._ZN13Kinect2Bridge10initializeEv[_ZN13Kinect2Bridge10initializeEv]+0x30c): undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'
kinect2_bridge.cpp:(.text._ZN13Kinect2Bridge10initializeEv[_ZN13Kinect2Bridge10initializeEv]+0x35a): undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'
kinect2_bridge.cpp:(.text._ZN13Kinect2Bridge10initializeEv[_ZN13Kinect2Bridge10initializeEv]+0x3ab): undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'
kinect2_bridge.cpp:(.text._ZN13Kinect2Bridge10initializeEv[_ZN13Kinect2Bridge10initializeEv]+0x3fc): undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'
kinect2_bridge.cpp:(.text._ZN13Kinect2Bridge10initializeEv[_ZN13Kinect2Bridge10initializeEv]+0x420): undefined reference to `ros::NodeHandle::getParam(std::string const&, std::string&) const'
kinect2_bridge.cpp:(.text._ZN13Kinect2Bridge10initializeEv[_ZN13Kinect2Bridge10initializeEv]+0x476): undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'
kinect2_bridge.cpp:(.text._ZN13Kinect2Bridge10initializeEv[_ZN13Kinect2Bridge10initializeEv]+0x4c7): undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'
kinect2_bridge.cpp:(.text._ZN13Kinect2Bridge10initializeEv[_ZN13Kinect2Bridge10initializeEv]+0x4eb): undefined reference to `ros::NodeHandle::getParam(std::string const&, std::string&) const'
kinect2_bridge.cpp:(.text._ZN13Kinect2Bridge10initializeEv[_ZN13Kinect2Bridge10initializeEv]+0x541): undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'
kinect2_bridge.cpp:(.text._ZN13Kinect2Bridge10initializeEv[_ZN13Kinect2Bridge10initializeEv]+0x592): undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'
kinect2_bridge.cpp:(.text._ZN13Kinect2Bridge10initializeEv[_ZN13Kinect2Bridge10initializeEv]+0x5e9): undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'
kinect2_bridge.cpp:(.text._ZN13Kinect2Bridge10initializeEv[_ZN13Kinect2Bridge10initializeEv]+0x640): undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'
kinect2_bridge.cpp:(.text._ZN13Kinect2Bridge10initializeEv[_ZN13Kinect2Bridge10initializeEv]+0x691): undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'
CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o:kinect2_bridge.cpp:(.text._ZN13Kinect2Bridge10initializeEv[_ZN13Kinect2Bridge10initializeEv]+0x6df): more undefined references to `ros::NodeHandle::hasParam(std::string const&) const' follow
CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o: In function `Kinect2Bridge::initialize()':
kinect2_bridge.cpp:(.text._ZN13Kinect2Bridge5startEv[_ZN13Kinect2Bridge5startEv]+0x574): undefined reference to `ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > const&, char const*, int, char const*)'
kinect2_bridge.cpp:(.text._ZN13Kinect2Bridge5startEv[_ZN13Kinect2Bridge5startEv]+0x631): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
collect2: error: ld returned 1 exit status
iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/build.make:245: recipe for target '/home/myubuntu/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge' failed
make[2]: *** [/home/myubuntu/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge] Error 1
CMakeFiles/Makefile2:1735: recipe for target 'iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/all' failed
make[1]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
CMakeFiles/kinect2_calibration.dir/src/kinect2_calibration.cpp.o: In function `Recorder::readImage(boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const>, cv::Mat&) const [clone .isra.101]':
CMakeFiles/kinect2_calibration.dir/src/kinect2_calibration.cpp.o: In function `CameraCalibration::calibrateIntrinsics(cv::Size_<int> const&, std::vector<std::vector<cv::Point3_<float>, std::allocator<cv::Point3_<float> > >, std::allocator<std::vector<cv::Point3_<float>, std::allocator<cv::Point3_<float> > > > > const&, std::vector<std::vector<cv::Point_<float>, std::allocator<cv::Point_<float> > >, std::allocator<std::vector<cv::Point_<float>, std::allocator<cv::Point_<float> > > > > const&, cv::Mat&, cv::Mat&, cv::Mat&, cv::Mat&, std::vector<cv::Mat, std::allocator<cv::Mat> >&, std::vector<cv::Mat, std::allocator<cv::Mat> >&)':
kinect2_calibration.cpp:(.text._ZN17CameraCalibration16storeCalibrationEv[_ZN17CameraCalibration16storeCalibrationEv]+0x62d): undefined reference to `ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > const&, char const*, int, char const*)'
collect2: error: ld returned 1 exit status
[... cut ...]
iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/build.make:507: recipe for target '/home/myubuntu/catkin_ws/devel/lib/kinect2_viewer/kinect2_viewer' failed
make[2]: *** [/home/myubuntu/catkin_ws/devel/lib/kinect2_viewer/kinect2_viewer] Error 1
CMakeFiles/Makefile2:2877: recipe for target 'iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/all' failed
make[1]: *** [iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed
`

ipgvl, 2019-02-01:
Please did you solve it?

asmand001, 2019-04-18:
Please did you solve it?