code-iai/iai_kinect2 #497: Error while calibrating depth

VasLem, 2018-07-29:
I have a rather peculiar setup here, which is Ubuntu 18.04, running a ROS melodic 1.14.2 and using OpenCV3.2 . I have to say that everything worked perfectly till now. I am facing this error while trying to calibrate depth image (`rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 calibrate depth`). The error is:

> OpenCV Error: Bad argument (Input argument is not a valid matrix) in cvRodrigues2, file /tmp/opencv/opencv-3.2.0/modules/calib3d/src/calibration.cpp, line 259
> terminate called after throwing an instance of 'cv::Exception'
>   what():  /tmp/opencv/opencv-3.2.0/modules/calib3d/src/calibration.cpp:259: error: (-5) Input argument is not a valid matrix in function cvRodrigues2

Does anyone know what is happening? I seem not to understand the source of the problem by looking at the project code.

anthony-simetri, 2020-11-18:
I am also having this problem.

thisisjaskaran, 2021-03-18:
Was this ever resolved?

sebastianzacarias, 2021-10-30:
Hi
I have the same problem
I tried using the command "record depth" but it is the same that "record sync"
I am using a hp laptop with AMD Ryzen
I do not know if the problem is generated by a possible bad config due to the use of AMD

gycn, 2021-12-16:
If you use OpenCV 4, lines 1093-1096 in kinect2_bridge/src/kinect2_calibration.cpp get macro'ed out, so you'll need to add a line for OpenCV 4 (same call signature as the one for OpenCV 3).

remmel, 2021-12-16:
> If you use OpenCV 4, lines 1093-1096 in kinect2_bridge/src/kinect2_calibration.cpp get macro'ed out, so you'll need to add a line for OpenCV 4 (same call signature as the one for OpenCV 3).

Thanks, I just came across that bug, and you've just answered it :) you saved my night

Rampantsen, 2023-04-15:
> If you use OpenCV 4, lines 1093-1096 in kinect2_bridge/src/kinect2_calibration.cpp get macro'ed out, so you'll need to add a line for OpenCV 4 (same call signature as the one for OpenCV 3).

Thanks a lot.