code-iai/iai_kinect2 #498: error: expected type-specifier for libfreenect

juliangaal, 2018-08-03:
Libreenect is installed and found by the cmake file in kinect2_bridge, however, during compilation i get this error:

```
/home/chitt/julian/reground_workspace/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘bool Kinect2Bridge::initPipeline(const string&, int32_t)’:
/home/chitt/julian/reground_workspace/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:427:28: error: expected type-specifier
       packetPipeline = new libfreenect2::OpenCLKdePacketPipeline(device);
                            ^
/home/chitt/julian/reground_workspace/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:436:28: error: expected type-specifier
       packetPipeline = new libfreenect2::CudaKdePacketPipeline(device);
                            ^
/home/chitt/julian/reground_workspace/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘bool Kinect2Bridge::initPipeline(const string&, int32_t)’:
/home/chitt/julian/reground_workspace/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:427:28: error: expected type-specifier
       packetPipeline = new libfreenect2::OpenCLKdePacketPipeline(device);
                            ^
/home/chitt/julian/reground_workspace/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:436:28: error: expected type-specifier
       packetPipeline = new libfreenect2::CudaKdePacketPipeline(device);
                            ^

```

That's strange, because libfreenect2 is used before that point in the same file. Any idea what the problem could be?

arnabsinha, 2018-08-23:
I am also facing the same problem

arnabsinha, 2018-08-23:
Solved it by doing a fresh installation as described here: https://github.com/LCAS/iliad/wiki/Installing-libfreenect2-and-iai_kinect2-in-Ubuntu-16.04