code-iai/iai_kinect2 #499: Packet buffer is NULL

frushton, 2018-08-03:
I have my kinectv2 plugged into a USB 3.0 port, and am getting the following. I get the repeating error "[DepthPacketStreamParser] Packet buffer is NULL", and I also get "[CudaDepthPacketProcessorImpl] cudaGetDeviceCount(&deviceCount): unknown error". I have tried googling, but that got me nowhere. Any thoughts?

... logging to /home/e4e/.ros/log/c71f040c-9764-11e8-88a2-9cb6d0177589/roslaunch-e4e-GS63VR-6RF-18319.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://e4e-GS63VR-6RF:36731/

SUMMARY
========

PARAMETERS
 * /imu_node/rate: 500
 * /imu_node/sensor_conf/6Deqbn/frame_id: imu_link
 * /imu_node/sensor_conf/6K7Fsi/frame_id: imu_link
 * /imu_node/sensor_conf/n3J/frame_id: base_ultrasonic
 * /imu_node/sensor_conf/n3J/max: 4.0
 * /imu_node/sensor_conf/n3J/min: 0.02
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/e4e/catkin_...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 8.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: True
 * /kinect2_bridge/queue_size: 5
 * /kinect2_bridge/reg_device: -1
 * /kinect2_bridge/reg_method: default
 * /kinect2_bridge/sensor: 
 * /kinect2_bridge/use_png: False
 * /kinect2_bridge/worker_threads: 4
 * /kinect2_points_xyzrgb_hd/queue_size: 5
 * /kinect2_points_xyzrgb_qhd/queue_size: 5
 * /kinect2_points_xyzrgb_sd/queue_size: 5
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /rtabmap/rgbd_odometry/approx_sync: True
 * /rtabmap/rgbd_odometry/config_path: 
 * /rtabmap/rgbd_odometry/frame_id: kinect2_link
 * /rtabmap/rgbd_odometry/ground_truth_base_frame_id: 
 * /rtabmap/rgbd_odometry/ground_truth_frame_id: 
 * /rtabmap/rgbd_odometry/odom_frame_id: odom
 * /rtabmap/rgbd_odometry/queue_size: 10
 * /rtabmap/rgbd_odometry/wait_for_transform_duration: 0.2
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/approx_sync: True
 * /rtabmap/rtabmap/config_path: 
 * /rtabmap/rtabmap/database_path: ~/.ros/rtabmap.db
 * /rtabmap/rtabmap/frame_id: kinect2_link
 * /rtabmap/rtabmap/ground_truth_base_frame_id: 
 * /rtabmap/rtabmap/ground_truth_frame_id: 
 * /rtabmap/rtabmap/map_frame_id: map
 * /rtabmap/rtabmap/odom_frame_id: 
 * /rtabmap/rtabmap/odom_sensor_sync: False
 * /rtabmap/rtabmap/odom_tf_angular_variance: 1.0
 * /rtabmap/rtabmap/odom_tf_linear_variance: 1.0
 * /rtabmap/rtabmap/queue_size: 10
 * /rtabmap/rtabmap/subscribe_depth: True
 * /rtabmap/rtabmap/subscribe_scan: False
 * /rtabmap/rtabmap/subscribe_scan_cloud: False
 * /rtabmap/rtabmap/subscribe_stereo: False
 * /rtabmap/rtabmap/subscribe_user_data: False
[... 101 lines cut ...]
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Info] [Freenect2DeviceImpl] started
[ INFO] [1533331974.257529078]: [Kinect2Bridge::initDevice] device serial: 169765140447
[ INFO] [1533331974.257556119]: [Kinect2Bridge::initDevice] device firmware: 4.0.3916.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL

alliansmover, 2018-10-19:
@frushton Hello my friend , I have the same problem.if  you solved it?

heyonghong, 2018-11-20:
hi my friend ,i have the same problem, are you solved it?

fmacrae, 2019-04-29:
Same here on ubuntu 16.04 with kinetic

steps to reproduce:

rosrun kinect2_viewer kinect2_viewer sd cloud

roslaunch kinect2_bridge kinect2_bridge.launch

fmacrae, 2019-04-29:
LOL it's the OpenCV intall being used by default

You can fix by switch to CPU or fix the openCV install to the version they like:

rosrun kinect2_bridge kinect2_bridge _depth_method:=opengl _reg_method:=cpu

fmacrae, 2019-04-29:
Check out this issue:  

https://github.com/code-iai/iai_kinect2/issues/371

cardboardcode, 2019-10-11:
## [Solution]

In issue #371, as highlighted by @fmacrae and @leviresende, 

the solution is to simply do the following:

> 1. Go to [`kinect2_bridge.launch`](https://github.com/code-iai/iai_kinect2/blob/master/kinect2_bridge/launch/kinect2_bridge.launch) under `kinect2_bridge` package.
> 2. Change `depth_method` parameter value to **opengl** in [line 12 of the file](https://github.com/code-iai/iai_kinect2/blob/master/kinect2_bridge/launch/kinect2_bridge.launch#L12).
> 3. Change `reg_method` parameter value to **cpu** in [line 14 of the file](https://github.com/code-iai/iai_kinect2/blob/master/kinect2_bridge/launch/kinect2_bridge.launch#L14).