code-iai/iai_kinect2 #500: catkin_make error

marooncn, 2018-08-04:
Hi, I install this package as your instruction. My system is Ubuntu16.04 and I have installed libfreenect2 successfully. When I finally run
`catkin_make -DCMAKE_BUILD_TYPE="Release"`
There is an error
> make[2]: *** No rule to make target '/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_calib3d3.so.3.3.1', needed by '/home/maroon/catkin_ws/devel/lib/kinect2_viewer/kinect2_viewer'。 停止。
CMakeFiles/Makefile2:5027: recipe for target 'iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/all' failed
make[1]: *** [iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

But I can still run 
`roslaunch kinect2_bridge kinect2_bridge.launch`
with the errors:

> [ERROR] [1533367777.295733944, 7408.775000000]: Skipped loading plugin with error: XML Document '/opt/ros/kinetic/share/gmapping/nodelet_plugins.xml' has no Root Element. This likely means the XML is malformed or missing..
[ERROR] [1533367777.299979435, 7408.777000000]: Failed to load nodelet [/kinect2_bridge] of type [kinect2_bridge/kinect2_bridge_nodelet] even after refreshing the cache: Could not find library corresponding to plugin kinect2_bridge/kinect2_bridge_nodelet. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1533367777.300163374, 7408.777000000]: The error before refreshing the cache was: Could not find library corresponding to plugin kinect2_bridge/kinect2_bridge_nodelet. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[FATAL] [1533367777.300997540]: Failed to load nodelet '/kinect2_bridge` of type `kinect2_bridge/kinect2_bridge_nodelet` to manager `kinect2'
[kinect2_bridge-2] process has died [pid 23005, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/home/maroon/.ros/log/20f660b2-97a4-11e8-9558-0242baa29cb4/kinect2_bridge-2.log].

How can I solve it? Thank you.

Magical-E, 2019-06-07:
I have the same error in jetson nano,was there any solution for this?

guo1104b, 2019-08-01:
老哥，我现在遇到了同样的问题，你当时是怎么解决的？