code-iai/iai_kinect2 #504: How can I get raw sd size color images?

tokyokuma, 2018-08-23:
I'm using KinectV2 on Jetson TX2.
Considering the processing speed of Jetson, I want to get SD size color image.
However, when I subscribed this topic (/kinect2/sd/image_color_rect), the acquired image is a color image combined with the depth image.

I believe that SD color images can be obtained if I can modify this file (kinect2_bridge.cpp). However, I tried several times on my own, but I could not do it well.

help me!!

SriGanesh130, 2018-09-18:
Same thing i am trying, if u find the solution remind me.
Thanks

ManavPoddarRobroSystems, 2019-02-12:
Did you guys found any solution? Please help !!!

tokyokuma, 2019-02-12:
Sorry,  I gave up before solving the problem.

ManavPoddarRobroSystems, 2019-02-12:
Did you use any alternate for this? I am using both the qhd image topics i.e. kinect2/qhd/image_color and kinect2/qhd/image_depth_rect. I am getting the desired results.

tokyokuma, 2019-02-12:
I process images acquired from Kinect using deep learning. I use qhd size, but I resize both RGB and depth image before doing image processing and make it small. At first, this method prevents processing speed reduction.

liuhang0727, 2019-04-11:
This need is highly required, help me please!!!!

liuhang0727, 2019-04-11:
> Did you use any alternate for this? I am using both the qhd image topics i.e. kinect2/qhd/image_color and kinect2/qhd/image_depth_rect. I am getting the desired results.

when I use the qhd size in ROS, the timestamp of received msgs between kinect2/qhd/image_depth_rect and kinect2/qhd/image_color differ 300ms!!  Do you konw why? Do you meet this problem?

bomtorazek, 2019-10-27:
> when I use the qhd size in ROS, the timestamp of received msgs between kinect2/qhd/image_depth_rect and kinect2/qhd/image_color differ 300ms!! Do you konw why? Do you meet this problem?

I think it is because rectification needs some time to be processed.
BTW, do you know the way to save both color and depth images at the same time??
I am currently working with this command in bash.rc "rosrun image_view image_saver image:=/kinect2/qhd/image_depth_rect _filename_format:=kinect_ros/save_image/depth/depth_%05d.png _encoding:=16UC1"

JinXiangLai, 2021-12-11:
I am also meeting this problem. I search the kinect2_bridge.cpp file carefully and I find out that the author never publish true SD color image. What a shame!