code-iai/iai_kinect2 #506: fatal error: pcl/point_cloud.h: No such file or directory

shadowdk3, 2018-10-02:
Hi, I use 
`catkin_make -DCMAKE_BUILD_TYPE="Release" -Dfreenect2_DIR=/usr/local/freenect2/lib/cmake/freenect2` 

and get 
```
[ 27%] Built target kinect2_registration
[ 36%] Building CXX object iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/src/viewer.cpp.o
[ 54%] Built target kinect2_calibration
[ 72%] Built target kinect2_bridge_nodelet
[ 90%] Built target kinect2_bridge
/home/user/catkin_ws/src/iai_kinect2/kinect2_viewer/src/viewer.cpp:29:29: fatal error: pcl/point_cloud.h: No such file or directory
compilation terminated.
```
I have already compile and install pcl 1.8.0, I can find the file `pcl/point_cloud.h` under `/usr/local/include/pcl-1.8/`

Any idea?

os: ubuntu 16.04
pcl: 1.8.0

YongLD, 2018-10-09:
Have you solve the problem with the iai_kinect2 when using pcl1.8? I would like to use pcl1.8 but there are a lot of problem.

bbferka, 2018-10-23:
@shadowdk3 have you tried printing the PCL_INCLUDE_DIRS from the Cmake? I don't see why it would not compile with 1.8; There are no breaking changes between 1.7 and 1.8;

hwLiu8, 2022-08-02:
you can add the pcl path(  /usr/local/include/pcl-1.8/  ) under include_directories in you CMakeList.txt, in my computer, solve this problem.