code-iai/iai_kinect2 #514: Unloading nodelet /kinect2_bridge from manager kinect2

Chunpeng19, 2018-11-09:
... logging to /home/lynn/.ros/log/5571e2f8-e3c1-11e8-9ec5-f44d30637e8e/roslaunch-lynn-desktop-3538.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://lynn-desktop:38313/

SUMMARY
========

PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/lynn/catkin...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 12.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: False
 * /kinect2_bridge/queue_size: 5
 * /kinect2_bridge/reg_device: -1
 * /kinect2_bridge/reg_method: default
 * /kinect2_bridge/sensor: 
 * /kinect2_bridge/use_png: False
 * /kinect2_bridge/worker_threads: 4
 * /kinect2_points_xyzrgb_hd/queue_size: 5
 * /kinect2_points_xyzrgb_qhd/queue_size: 5
 * /kinect2_points_xyzrgb_sd/queue_size: 5
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    kinect2 (nodelet/nodelet)
    kinect2_bridge (nodelet/nodelet)
    kinect2_points_xyzrgb_hd (nodelet/nodelet)
    kinect2_points_xyzrgb_qhd (nodelet/nodelet)
    kinect2_points_xyzrgb_sd (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [3548]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 5571e2f8-e3c1-11e8-9ec5-f44d30637e8e
process[rosout-1]: started with pid [3561]
started core service [/rosout]
process[kinect2-2]: started with pid [3576]
process[kinect2_bridge-3]: started with pid [3579]
process[kinect2_points_xyzrgb_sd-4]: started with pid [3580]
[ INFO] [1541728014.425181243]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1541728014.431834153]: waitForService: Service [/kinect2/load_nodelet] has not been advertised, waiting...
process[kinect2_points_xyzrgb_qhd-5]: started with pid [3586]
process[kinect2_points_xyzrgb_hd-6]: started with pid [3599]
[ INFO] [1541728014.453286723]: Initializing nodelet with 4 worker threads.
[ INFO] [1541728014.475617661]: waitForService: Service [/kinect2/load_nodelet] is now available.
[ INFO] [1541728014.535949275]: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/lynn/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: default
     depth_device: -1
       reg_method: default
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 5
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) HD Graphics 6000 BroadWell U-Processor GT3 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Intel(R) HD Graphics 6000 BroadWell U-Processor GT3 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_39
libva info: va_openDriver() returns 0
[Info] [VaapiRgbPacketProcessorImpl] driver: Intel i965 driver for Intel(R) Broadwell - 1.7.0
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 8 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @3:2 with serial 035952434347
[Info] [Freenect2Impl] found 1 devices
[ INFO] [1541728015.149309701]: [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [1541728015.149355191]: [Kinect2Bridge::initDevice]   0: 035952434347 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [1541728015.305319113]: [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [1541728015.664821589]: [Kinect2Bridge::initDevice] device serial: 035952434347
[ INFO] [1541728015.664935429]: [Kinect2Bridge::initDevice] device firmware: 4.0.3917.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[ WARN] [1541728016.263573505]: [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [1541728016.263640859]: [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [1541728016.263683249]: [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [1541728016.263713775]: [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ INFO] [1541728016.296849545]: [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [1541728016.296936194]: [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [1541728016.303854065]: [DepthRegistrationOpenCL::init] devices:
[ INFO] [1541728016.303919612]: [DepthRegistrationOpenCL::init]   0: Intel(R) HD Graphics 6000 BroadWell U-Processor GT3
[ INFO] [1541728016.303960350]: [DepthRegistrationOpenCL::init] selected device: Intel(R) HD Graphics 6000 BroadWell U-Processor GT3
[ INFO] [1541728016.408762444]: [DepthRegistrationOpenCL::init] devices:
[ INFO] [1541728016.408811727]: [DepthRegistrationOpenCL::init]   0: Intel(R) HD Graphics 6000 BroadWell U-Processor GT3
[ INFO] [1541728016.408843086]: [DepthRegistrationOpenCL::init] selected device: Intel(R) HD Graphics 6000 BroadWell U-Processor GT3
[ INFO] [1541728016.531166047]: [Kinect2Bridge::main] waiting for clients to connect
^C[kinect2_points_xyzrgb_hd-6] killing on exit
[kinect2_points_xyzrgb_qhd-5] killing on exit
[kinect2_points_xyzrgb_sd-4] killing on exit
[kinect2-2] killing on exit
[kinect2_bridge-3] killing on exit
[ INFO] [1541728019.238412465]: Unloading nodelet /kinect2_bridge from manager kinect2
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Chunpeng19, 2018-11-09:
I have followed the step of instruction. But I cannot run kinect2_bridge correctly. It was always waiting for clients to connect. The only problem is that I am using ROS Kinetic here instead of Indigo. I'm curious if this bridge can run on Kinetic. If so, how could I solve my current problem?

pcoenen, 2018-11-09:
It is normal that it is waiting for clients to connect. 

Run the following command in another terminal window to see the camera output

`rosrun kinect2_viewer kinect2_viewer kinect2 sd cloud`

Chunpeng19, 2018-11-09:
Thanks so much!

closed 2018-11-09 by Chunpeng19