code-iai/iai_kinect2 #523: Invoking "make -j8 -l8" failed

sunjeet95, 2019-02-09:
Getting the following error for catkin_make -DCMAKE_BUILD_TYPE="Release"

In file included from /home/sanjeevs/catkin_ws/src/iai_kinect2/kinect2_viewer/src/viewer.cpp:32:0:
/usr/include/pcl-1.8/pcl/visualization/cloud_viewer.h:202:14: warning: ‘template<class> class std::auto_ptr’ is deprecated [-Wdeprecated-declarations]
         std::auto_ptr<CloudViewer_impl> impl_;
              ^~~~~~~~
In file included from /usr/include/c++/7/memory:80:0,
                 from /usr/include/c++/7/thread:39,
                 from /home/sanjeevs/catkin_ws/src/iai_kinect2/kinect2_viewer/src/viewer.cpp:26:
/usr/include/c++/7/bits/unique_ptr.h:51:28: note: declared here
   template<typename> class auto_ptr;
                            ^~~~~~~~
/home/sanjeevs/catkin_ws/src/iai_kinect2/kinect2_viewer/src/viewer.cpp: In member function ‘void Receiver::imageViewer()’:
/home/sanjeevs/catkin_ws/src/iai_kinect2/kinect2_viewer/src/viewer.cpp:279:84: error: ‘CV_AA’ was not declared in this scope
         cv::putText(combined, oss.str(), pos, font, sizeText, colorText, lineText, CV_AA);
                                                                                    ^~~~~
/home/sanjeevs/catkin_ws/src/iai_kinect2/kinect2_viewer/src/viewer.cpp:279:84: note: suggested alternative: ‘CV_C’
         cv::putText(combined, oss.str(), pos, font, sizeText, colorText, lineText, CV_AA);
                                                                                    ^~~~~
                                                                                    CV_C
iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/build.make:62: recipe for target 'iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/src/viewer.cpp.o' failed
make[2]: *** [iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/src/viewer.cpp.o] Error 1
CMakeFiles/Makefile2:2863: recipe for target 'iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/all' failed
make[1]: *** [iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

cshreyastech, 2019-06-02:
I have the same error in TX2, was there any solution for this?

Thanks

fengjiufa, 2019-11-30:
Me too . please tell me how to solve it.
Thanks!

Vgabond, 2019-12-12:
I have the same problem. I tried every means and follow the step to install dependencies. However, it still didn't work. How you solve the problem? Thanks very much!