code-iai/iai_kinect2 #528: Does Point cloud give real x,y,z?

chandanlal92, 2019-03-28:
I have a small doubt just for the clarification. I know this has been discussed before. I am finding the pose of some object with respect it kinect2 sensor frame. I am able to map the pixel values located to the x,y,z in the point cloud stream. My doubt is this point cloud data (x,y,z) measurement is true world coordinates or should I still have to do some transformations from calibration data?

kevinglasson, 2019-04-18:
Off topic: How were you able to map the pixel values to the x, y, z in the point cloud?

chandanlal92, 2019-04-19:
I am mapping the pixel using the point cloud data[index]
index=x+y*cloud->width;
cloud[index].x=x;
cloud[index].y=y;
cloud[index].z=z;
My pixels are mapping exactly to the point cloud but the problem is when I try to map the pixel using intrinsic parameters I could not map it in the point cloud .
I just need to know with what intrinsic formula qhd/points created.