code-iai/iai_kinect2 #532: Failed to load nodelet '/kinect2_bridge` of type `kinect2_bridge/kinect2_bridge_nodelet` to manager `kinect2'

Little-Potato-1990, 2019-04-22:
Hi
When I run "roslaunch kinect2_bridge kinect2_bridge.launch", I get the error "Failed to load nodelet '/kinect2_bridge` of type `kinect2_bridge/kinect2_bridge_nodelet` to manager `kinect2'"
Then I run it with gdb, I get the message: "Thread 1 "kinect2_bridge" received signal SIGSEGV, Segmentation fault. 0x00007fffccdabbf0 in llvm::cl::AddLiteralOption(llvm::cl::Option&, llvm::StringRef) () from /usr/lib/x86_64-linux-gnu/libLLVM-6.0.so.1"
Could you tell me how to fix it?
Thanks very much!

siehlema, 2019-05-24:
Same issue using the [OpenPTrack Docker](https://github.com/OpenPTrack/open_ptrack_v2/wiki/Docker-Images#single_camera_tracking)
Also related to this [Issue](https://github.com/OpenPTrack/open_ptrack_v2/issues/62)

siehlema, 2019-05-24:
I solved my issue by checking the Problems stated [here](https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting)

In my case I had to add a rules file (90-kinect2.rules into /etc/udev/rules.d) to get the needed access of the USB device.

zheng-sicheng, 2020-06-19:
same problem

arun-venkat-23, 2023-06-21:
Hey @siehlema! I'm encountering the same problem. Where did you add the rules file?

Also what was your ros and ubuntu version?