code-iai/iai_kinect2 #539: kinect and astra cannot be launched together nicely

torifujinami, 2019-07-23:
Hi,

I am working on a robot that has both a kinect v2 and an orbbec astra_mini. Unfortunately, we are running into this somewhat strange issue where if the kinect is running and we launch the astra mini, the astra mini is unable to output any data onto its topics. However, if launched in reverse order, if the astra mini is running and we launch the kinect, both sensors seem to run successfully. We are hoping that we can get to the bottom of this so that in the long term, the launching of the sensors is not coupled. 

Do you have any ideas why this might happen or know of anything we can try to get both to launch successfully independent of each other?

Thanks in advance!

JimmyDaSilva, 2019-07-23:
Hi, did you plug the two sensors on two different USB bus?
There is probably two much data for only one bus

torifujinami, 2019-07-23:
> Hi, did you plug the two sensors on two different USB bus?
> There is probably two much data for only one bus

Hi Jimmy, thanks for commenting. I forgot to mention, but we did verify the devices are on different buses. The Kinect is currently on bus 4 and the astra_mini is currently on bus 3.