code-iai/iai_kinect2 #545: OpenCV version issue /  ROS Melodic

ukachyuthan, 2019-09-17:
I have ROS Melodic and am trying to build the workspace. I am not able to compile in ROS Melodic due to OpenCV errors. Melodic comes with OpenCV 4.0. Is there a version of this package that works with Melodic. 

Thanks.

TomasMerva, 2019-10-13:
It is working with ROS Melodic. So far I do not have enough experience with this library, although I was able to install and launch `roslaunch kinect2_bridge kinect2_bridge.launch` without any problem. I have been using OpenCV 3.4.7, which is the newest version of OpenCV 3.

fengjiufa, 2019-11-30:
hello, please tell me how to install kinect2 truely. I can't do it sucessful.

SiddhantNadkarni, 2020-04-01:
@TomasMerva Could you please share how did you make the installation compatible with OpenCV Version 3.4.7?

pradhanshrijal, 2020-04-17:
I followed the instructions provide. Everything was working well until I launched `roslaunch kinect2_bridge kinect2_bridge.launch`. I got an issue similar to [this](https://github.com/code-iai/iai_kinect2/issues/489). I was wondering if it was because I have not calibrated the camera because the error occurs in the rgb data from the camera. The [libfreenect2](https://github.com/OpenKinect/libfreenect2) package is working fine.

I am using OpenCV 3.2 for Melodic and OpenCV 2.4.13.7 is also installed in my system.

Any advice would be welcomed.